nebrius/raspi-io #88: How to use Johnny-five in Angular2 project

lapereirabr, 2017-04-04:
Please, I am new to angular2 and would like help implementing Johnny-Five in angular2.

1 - I created a provider (J5Provider):

```
import { Injectable } from '@angular/core';
import 'rxjs/add/operator/map';

var raspi = require('raspi-io');
var j5 = require('johnny-five');

@Injectable()
export class J5Provider {

  private board: any;

  constructor() {
    this.board = new j5.Board({
      io: new raspi()
    });

    this.board.on('ready', function () {
      var led = new j5.Led('P1-7');
      led.strobe(500);
    });
  }
}
```
2 - Then I tried to instantiate in app.module.ts in an Ionic2 project:

import { NgModule, ErrorHandler } from '@angular/core';
import { IonicApp, IonicModule, IonicErrorHandler } from 'ionic-angular';
import { MyApp } from './app.component';
import { AboutPage } from '../pages/about/about';
import { ContactPage } from '../pages/contact/contact';
import { HomePage } from '../pages/home/home';
import { TabsPage } from '../pages/tabs/tabs';

import { StatusBar } from '@ionic-native/status-bar';
import { SplashScreen } from '@ionic-native/splash-screen';

import {J5Provider} from '../providers/j5-provider'

@NgModule({
  declarations: [
    MyApp,
    AboutPage,
    ContactPage,
    HomePage,
    TabsPage
  ],
  imports: [
    IonicModule.forRoot(MyApp)
  ],
  bootstrap: [IonicApp],
  entryComponents: [
    MyApp,
    AboutPage,
    ContactPage,
    HomePage,
    TabsPage
  ],
  providers: [
    StatusBar,
    SplashScreen,
    J5Provider,
    {provide: ErrorHandler, useClass: IonicErrorHandler}
  ]
})
export class AppModule {}
3 - But, when I run I get the following error:

repl.js:1 Uncaught Error: Cannot find module "repl"
    at v (polyfills.js:3)
    at webpackMissingModule (repl.js:1)
    at Object.<anonymous> (repl.js:1)
    at Object.<anonymous> (repl.js:106)
    at __webpack_require__ (bootstrap c6de802…:19)
    at Object.<anonymous> (board.js:19)
    at Object.<anonymous> (board.js:1269)
    at __webpack_require__ (bootstrap c6de802…:19)
    at Object.<anonymous> (accelerometer.js:1)
    at Object.<anonymous> (accelerometer.js:1175)
    at __webpack_require__ (bootstrap c6de802…:19)
    at Object.noop (johnny-five.js:11)
    at __webpack_require__ (bootstrap c6de802…:19)
    at Object.<anonymous> (app.component.ts:8)
    at __webpack_require__ (bootstrap c6de802…:19)
Please can anyone guide me on how I do this implementation? Thanks in advance.

tralves, 2017-04-04:
Hello, @lapereirabr!

What you are trying to do is impossible. Johnny-five runs with NodeJS, on the Raspberry-Pi.
Angular runs on the browser. That node.js stuff you are using (npm, webpack, etc) will only build the Angular app that will then run on the browser.

So, how can you do it? You will have to build a Node.JS app that uses Johnny-five to work with the Raspberry-Pi hardware. This NodeJS app will have to expose an API (REST API, socket.io, or something else).

Then you will build the interface app with Angular. This app will consume the API the NodeJS exposes.

I built something similar some time ago: https://github.com/tralves/j5-pi2go

That project includes a Node.JS server to control the Raspberry-PI robot and exposes a socket.io API.

Inside, there is a `j5-pi2go-ui` folder which is a VueJS (could be Angular) app with the UI. This app connects to the node.js server through socket.io.

Sorry for the bad news :)

nebrius, 2017-04-04:
Depending on the browser environment you want to target, you _might_ be able to use [Page Nodes](https://github.com/monteslu/pagenodes), but as @tralves said, this project is Node.js only.

lapereirabr, 2017-04-05:
Hi, friends.

I thought about this path after seeing this project. (https://github.com/urish/ng2-simon)
I tried to implement something simpler based on it.
Anyway, thank you for your answers, they were very enlightening and gave me an idea of what path to take in my project.

nebrius, 2017-04-05:
Best of luck @lapereirabr! I'm going to close this issue, but feel free to reopen if there's more to discuss!

closed 2017-04-05 by nebrius