nebrius/raspi-io #91: GPIO is broken

nebrius, 2017-04-21:
I don't know why, but the raspi-io is broken now. Pin modes are not being set correctly via calls to `pinMode()` in the C++ side of things within raspi-gpio. I've been investigating to see if a new version of Wiring Pi caused it, but that doesn't _appear_ to be the case.

Unfortunately, I'm completely stumped why this is happening. I'll keep digging though. Worse comes to worse, I'll dump Wiring Pi completely and switch to pigpio for everything, which is already used in part of Raspi.js

andrewn, 2017-04-23:
That sounds really annoying. Let me know if there's anything I can do to lend a hand.

nebrius, 2017-04-26:
Whew, Raspi IO 8.0.0 is out! It should fix this issue because we no longer rely on Wiring Pi!

@andrewn, if you have a chance, could you take 8.0.0 for a spin and make sure it still works for you?

andrewn, 2017-05-07:
I've just tried Raspi IO 8.0.1 on a Pi Zero W with one of my PCBs that has a few RGB LEDs (soft PWM) and push-buttons. It's all working fine for me.

Thanks so much for the rapid rewrite!

nebrius, 2017-05-09:
Thanks for testing @andrewn! I'm going to go ahead and close this now, I think we've managed to find all the regressions.

closed 2017-05-09 by nebrius