nebrius/raspi-io #92: Issue installing raspi-io

myfiaf, 2017-04-21:
Hello,

I am trying to install raspi-io on my raspberry pi 2 running raspbian Jessie.
I installed nodejs (v0.10.29), node (v4.2.1) and npm (4.5.0).
When I run npm install raspi-io, I got error messages (I put the log in attachment).

Could anyone help me?

Olivier

[2017-04-21T18_33_18_581Z-debug.log.txt](https://github.com/nebrius/raspi-io/files/948195/2017-04-21T18_33_18_581Z-debug.log.txt)

nebrius, 2017-04-21:
Can you explain why you installed two versions of Node.js, and which version you're using to run your app? Node v0.10 is not supported by Raspi IO, but Node v4 is.

myfiaf, 2017-04-21:
ok, my bad... I had not wiringpi installed...
after installing it, installing raspi-io worked...

sorry for the trouble

Olivier

nebrius, 2017-04-21:
No trouble at all, I'm glad you got it working! :)

closed 2017-04-21 by nebrius