nebrius/raspi-io #93: Running script after boot up

WillemMe, 2017-04-26:
Hi,
I'm working a camera project that make use of node.js to take pictures and to handler a web server.
But for it to work it needs the script to be run after boot up.
At the moment if have tried to run it to rc.local with:
`su pi -c '/usr/local/bin/node /home/pi/piSight/app.js'`
I also tried by making a init.d script.
They both "work" but after they launch the board closes independently, so no good.

I'm I doing something wrong or is it impossible to run a script on boot up

Thanks in advance

Willem Me

nebrius, 2017-04-28:
Can you give me some more details about this project?

- Are you using Raspi IO with Johnny-Five for this project? If so, what are you using them for? (Neither Raspi IO nor Johnny-Five support cameras nor web servers, those are handled by other packages)
- What is the `pi` command?
- What is the contents of `app.js`?
- Can you explain what you mean by "the board closes independently"?

WillemMe, 2017-04-29:
I'm sorry I was a bit unclear, I will break it down.

The core idea is making a multifunctioning camera, with features like time lapsing, 3sec GIFs, movies and stills. As you mentioned Johnny-Five neither Raspi IO deals with camera operations, another package does this [raspicam](https://www.npmjs.com/package/raspicam). Johnny-Five is used for the user input via buttons and is used for controlling the LCD (16 X 2).
The buttons are used to scroll through the menu displayed on the LCD, to enter Sub menu and start camera operations.
If my understanding is right, to use Johnny-Five on a Raspberry pi without an Arduino you need Raspi IO, that is where that comes to play.
 
### The code:
The code found [here](https://github.com/WillemMe/PiSight/blob/master/app.js) runs the operation.
It consists out of multiple parts:
- Setup
- Streaming handler with SocketIO and express
- Input handler (Johnny-Five part)
- Camera setup (raspicam)
- LCD displaying, and menu objects (mix)

### The problem:
the "pi command": was a bit redundant because just running  `/usr/local/bin/node /home/pi/piSight/app.js` would do the same. the command is just `node app.js`. No sudo because every thing in rc.local is ran by root.(If you are unfamiliar with rc.local: it just runs command on boot)
When this command is ran normally (in a ssh terminal), you get the: 
```
Available RaspberryPi-IO
Connected RaspberryPi-IO
Repl Initialized
```
And then it waits for inputs, and only closes by SIGINT by pressing ctrl-c twice.
and then again:
`Board Closing.`

But here comes the problem:
When rc.local runs the command you get your, Available, connected and Repl Initialized.
But then a second afterwards you get `Board Closing.` as if you pressed ctrl-c.
This is what I meant with: "the board closes independently".
I have tried differed things:
[like this one]( http://stackoverflow.com/questions/29069192/node-js-on-boot)
Or the same as [this person]( https://github.com/frbrkoala/node-pi-softshutdown)
But had no luck.
I think it has something to do with the rc.local finishing and telling de board it is done.
Do you have any experience with running it on boot/ on start up.
I’m I just forgetting something or is it just impossible.
Willem Me

 hopefully I didn't make it tl;dr

nebrius, 2017-04-29:
Oh ok, got it. So here's what I think is happening: IIRC, init.d scripts have timeouts associated with them, and if they don't exit in time, the OS kills it (which would explain the "board closing" error you're seeing).

What you'll need to do is run the script in the background with the following (note the `&`):

```Bash
/usr/local/bin/node /home/pi/piSight/app.js &
```

Alternatively, you can create a [systemd service](https://www.digitalocean.com/community/tutorials/how-to-deploy-node-js-applications-using-systemd-and-nginx) or use [pm2](http://pm2.keymetrics.io/) if you want something a little more fully featured.

WillemMe, 2017-04-29:
Thanks for the quick reply,
no luck with adding the & to rc.local, the board is still closing.
I made a systemd
```
Description=PiSight NodeJS

[Service]
ExecStart=/usr/local/bin/node /home/pi/piSight/app.js
#WorkingDirectory=/home/pi/piSight   # Required on some systems
Restart=always
RestartSec=10                       # Restart service after 10 seconds if node service crashes
StandardOutput=syslog               # Output to syslog
StandardError=syslog                # Output to syslog
SyslogIdentifier=Pi Sight
#User=<alternate user>
#Group=<alternate group>
Environment=NODE_ENV=production PORT=80

[Install]
WantedBy=multi-user.target

```
 But it still closses and restarts as it is supposed to because of the service.
evidence can be found in the systemlog:
```
Apr 29 23:37:14 PiSight systemd[1]: Starting pisight.service...
Apr 29 23:37:14 PiSight systemd[1]: Started pisight.service.
Apr 29 23:37:27 PiSight Pi Sight[1880]: 1493501847306 Available RaspberryPi-IO
Apr 29 23:37:27 PiSight Pi Sight[1880]: 1493501847485 Connected RaspberryPi-IO
Apr 29 23:37:27 PiSight Pi Sight[1880]: 1493501847530 Repl Initialized
Apr 29 23:37:28 PiSight Pi Sight[1880]: >> 1493501848530 Board Closing.
Apr 29 23:37:28 PiSight systemd[1]: pisight.service holdoff time over, scheduling restart.
Apr 29 23:37:28 PiSight systemd[1]: Stopping pisight.service...
Apr 29 23:37:28 PiSight systemd[1]: Starting pisight.service...
Apr 29 23:37:28 PiSight systemd[1]: Started pisight.service.
Apr 29 23:37:41 PiSight Pi Sight[1906]: 1493501861565 Available RaspberryPi-IO
Apr 29 23:37:41 PiSight Pi Sight[1906]: 1493501861743 Connected RaspberryPi-IO
Apr 29 23:37:41 PiSight Pi Sight[1906]: 1493501861788 Repl Initialized
Apr 29 23:37:42 PiSight Pi Sight[1906]: >> 1493501862783 Board Closing.
```
I did find that this problem only occurs when trying to run a script with Johnny-Five and setting up a board. Because when it tried the exact same but then with a express server everything just worked out fine, no restarting. Also tried to run this one with a & after the exec command still no success.

My script self does not have any effect on this restarting, because with an empty script and just a board setup it still occurs.

The only suggestion I have now is that it has something to do with my installment of Raspi-IO.
So maybe could you recreate this problem with just a empty board setup it may reveal the answer.

Once again thank you,
Willem Me

WillemMe, 2017-05-13:
It has been some time,
I have tried some futher debugging with no success.
It has nothing to do with my installation of raspi-io because the same occures on a freshly installed Raspberry pi B 3.
I guess i'm out of luck

nebrius, 2017-05-13:
I'm sorry to hear you couldn't get it to work. Unfortunately I'm out of ideas too, other than to try PM2. I'm not an ops person, so this stuff is mostly outside my area of expertise. :(

WillemMe, 2017-05-13:
Success,
After many failed attempts it finally worked out.
By using the earlier mentioned systemd script but then replacing de node part of the command with foreverjs. [Forever](https://github.com/foreverjs/forever) is similar to PM2 but a bit easier.
`ExecStart=/usr/local/bin/forever /home/pi/piSight/app.js`
Thanks a lot for all the help, I really appreciate it!
I suggest adding a note so future developers don’t have to go through this frustration struggler.
Willem Me out.

nebrius, 2017-05-14:
I'm glad you got it to work! I'm going to go ahead and close this issue, feel free to reopen if need be.

closed 2017-05-14 by nebrius

npsm, 2017-06-12:
I ran into the same problem and found that setting repl when constructing the board object solved the issue. eg: var Board = new five.Board({repl: false});
It appears that repl isn't handling the linux signals that well, systemd uses these to determine if a service is started or not.
Hope this help someone running into the same issue.

wrodrigues7, 2020-07-04:
@WillemMe , I really appreciate it.
It's all I needed to finish my IoT project. Thank you for your help.

By the way, I'm seeing it in 2020.

Regards,

Wesley Rodrigues