nebrius/raspi-io #94: Pin Mode Error - Pi3, GPS GP-20U7 & Johnny-Five

avcohen, 2017-05-28:
I used `raspi-config` to disable Serial/USB Console and Enabled Serial interface. I've also added `enable_uart=1` config.txt.

Running `cat /dev/serial0` from the command like outputs a stream of data from the GPS unit and seems to be doing it's job. Nonetheless, I'm having an issue with Pin 10 receiving data from the GPS Unit. When I run my code below it throws the error:

> Error: Pin "10" does not support mode "other"
  	 	

    var five = require("johnny-five");
    var Raspi = require("raspi-io");
    var board = new five.Board({
      io : new Raspi({ enableSerial : true /* apparently required as per raspi-io docs */ })
    });
    
        board.on("ready", function() {
            const gps = new five.GPS({
                pins : {
                        rx : 'P1-10',
                        tx : 'P1-8',
                }
        });

        gps.on("data", () => {
                console.log("position");
                console.log(" latitude  : ", this.latitude);
                console.log(" longitude : ", this.longitude);
                console.log(" altitude  : ", this.altitude);
                console.log("———————————————————————————————————");
        });

    });

nebrius, 2017-05-30:
Try this:

```
const five = require("johnny-five");
const Raspi = require("raspi-io");
const board = new five.Board({
    io : new Raspi({ enableSerial: true /* apparently required as per raspi-io docs */ })
});

board.on("ready", () => {
    const gps = new five.GPS({});

    gps.on("data", () => {
        console.log("position");
        console.log(" latitude  : ", gps.latitude);
        console.log(" longitude : ", gps.longitude);
        console.log(" altitude  : ", gps.altitude);
        console.log("———————————————————————————————————");
    });
});
```

I did a few things here. Most important is that I removed the `pins` entry from the GPS constructor. The `pins` entry is really only useful if you need to used software emulated serial, something that isn't supported on the Raspberry Pi. Without `pins`, it defaults to using `port`, which itself defaults to an internal value supplied by Raspi IO.

Another thing I did, I changed from using `this.latitude` et al in the `data` event, because the arrow function causes the `this` value to be `global` instead of `gps`, which would make `this.latitude` to equal `undefined` in your code.

Let me know if you run into any other issues!

avcohen, 2017-06-02:
Thanks for the feedback this was a huge help. Big oversight on my part with the fat arrow and `this`. While this runs, the values returned are all `0`.

running either `cat /dev/ttyAMA0` or `cat /dev/serial0` outputs the following:  

`$GPGLL,,,,,,V,N*64
$GPRMC,,V,,,,,,,,,,N*53
$GPVTG,,,,,,,,,N*30
$GPGGA,,,,,,0,00,99.99,,,,,,*48
$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30`

so while I know data is coming through so there is data but it doesn't seem to be found in the latitude longitude and altitude objects of gps.

Thanks again.

nebrius, 2017-06-05:
Hmm, at this point, I would _guess_ that there is some sort of issue with parsing data somewhere in the stack.

Which GPA device are you using?

avcohen, 2017-06-05:
I'm using [GPS Receiver - GP-20U7 (56 Channel)](https://www.sparkfun.com/products/13740).

nebrius, 2017-06-07:
I've ordered one of these devices, I'll test it out as soon as it arrives

nebrius, 2017-06-15:
ok, I did some testing and I confirmed that Raspi IO is outputting what _looks_ like valid data, but the GPS parser in Johnny-Five isn't able to parse it.

@rwaldron, do you see anything fishy about this output from Raspi IO?

[image]

```
,,,,,,,,N*30\r\n$GPGGA,,,,,,0,00,99.99,,,,,,*48\r\n$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30\r\n$GPGSV,3,1,09,02,,,09,03,,,09,04,,,16,05,,,16*72\r\n$GPGSV,3,2,09,08,,,24,16,,,08,24,,,26,26,,,24*70\r\n$GPGSV,3,3,09,27,,,24*73\r\n$GPGLL,,,,,,V,N*64\r\n$GPRMC,,V,,,,,,,,,,N*53\r\n$GPVTG,,,,,,,,,N*30\r\n$GPGGA,,,,,,0,00,99.99,,,,,,*48\r\n$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30\r\n$GPGSV,2,1,06,04,,,21,05,,,14,08,,,24,24,,,25*74\r\n$GPGSV,2,2,06,26,,,24,27,,,24*7E\r\n$GPGLL,,,,,,V,N*64\r\n$GPRMC,,V,,,,,,,,,,N*53\r\n$GPVTG,,,,,,,,,N*30\r\n$GPGGA,,,,,,0,00,99.99,,,,,,*48\r\n$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30\r\n$GPGSV,2,1,05,04,,,21,05,,,12,24,,,23,26,,,25*7A\r\n$GPGSV,2,2,05,27,,,25*7E\r\n$GPGLL,,,,,,V,N*64\r\n$GPRMC,,V,,,,,,,,,,N*53\r\n$GPVTG,,,,,,,,,N*30\r\n$GPGGA,,,,,,0,00,99.99,,,,,,*48\r\n$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30\r\n$GPGSV,2,1,05,04,,,18,08,,,28,24,,,20,26,,,26*74\r\n$GPGSV,2,2,05,27,,,26*7D\r\n$GPGLL,,,,,,V,N*64\r\n$GPRMC,,V,,,,,,,,,,N*53\r\n$GPVTG,,,,,,,,,N*30\r\n$GPGGA,,,,,,0,00,99.99,,,,,,*48\r\n$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30\r\n$GPGSV,2,1,06,04,,,15,08,,,27,10,,,26,24,,,17*74\r\n$GPGSV,2,2,06,26,,,26,27,,,25*7D\r\n$GPGLL,,,,,,V,N*64\r\n$GPRMC,,V,,,,,,,,,,N*53\r\n$GPVTG,,,,,,,,,N*30\r\n$GPGGA,,,,,,0,00,99.99,,,,,,*48\r\n$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30\r\n$GPGSV,2,1,05,08,,,26,10,,,26,24,,,13,26,,,26*72\r\n$GPGSV,2,2,05,27,,,26*7D\r\n$GPGLL,,,,,,V,N*64\r\n$GPRMC,,V,,,,,,,,,,N*53\r\n$GPVTG,,,,,,,,,N*30\r\n$GPGGA,,,,,,0,00,99.99,,,,,,*48\r\n$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30\r\n$GPGSV,2,1,05,08,,,26,10,,,26,24,,,13,26,,,26*72\r\n$GPGSV,2,2,05,27,,,26*7D\r\n$GPGLL,,,,,,V,N*64\r\n$GPRMC,,V,,,,,,,,,,N*53\r\n$GPVTG,,,,,,,,,N*30\r\n$GPGGA,,,,,,0,00,99.99,,,,,,*48\r\n$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30\r\n$GPGSV,2,1,06,08,,,26,10,,,26,11,,,26,24,,,15*73\r\n$GPGSV,2,2,06,26,,,26,27,,,26*7E\r\n$GPGLL,,,,,,V,N*64\r\n$GPRMC,,V,,,,,,,,,,N*53\r\n$GPVTG,,,,,,,,,N*30\r\n$GPGGA,,,,,,0,00,99.99,,,,,,*48\r\n$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30\r\n$GPGSV,2,1,07,08,,,25,10,,,26,11,,,25,13,,,28*78\r\n$GPGSV,2,2,07,24,,,15,26,,,26,27,,,26*7D\r\n$GPGLL,,,,,,V,N*64\r\n$GPRMC,,V,,,,,,,,,,N*53\r\n$GPVTG,,,,,,,,,N*30\r\n$GPGGA,,,,,,0,00,99.99,,,,,,*48\r\n$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30\r\n$GPGSV,2,1,06,08,,,26,10,,,26,11,,,24,24,,,15*71\r\n$GPGSV,2,2,06,26,,,26,27,,,26*7E\r\n$GPGLL,,,,,,V,N*64\r\n$GPRMC,,V,,,,,,,,,,N*53\r\n$GPVTG,,,,,,,,,N*30\r\n$GPGGA,,,,,,0,00,99.99,,,,,,*48\r\n$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30\r\n$GPGSV,2,1,05,08,,,26,10,,,26,11,,,23,26,,,26*77\r\n$GPGSV,2,2,05,27,,,26*7D\r\n$GPGLL,,,,,,V,N*64\r\n$GPRMC,,V,,,,,,,,,,N*53\r\n$GPVTG,,,,,,,,,N*30\r\n$GPGGA,,,,,,0,00,99.99,,,,,,*48\r\n$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30\r\n$GPGSV,1,1,04,08,,,26,10,,,26,11,,,23,27,,,26*74\r\n$GPGLL,,,,,,V,N*64\r\n$GPRMC,,V,,,,,,,,,,N*53\r\n$GPVTG,,,,,,,,,N*30\r\n$GPGGA,,,,,,0,00,99.99,,,,,,*48\r\n$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30\r\n$GPGSV,2,1,06,08,,,26,10,,,25,11,,,23,14,,,25*75\r\n$GPGSV,2,2,06,26,,,27,27,,,26*7F\r\n$GPGLL,,,,,,V,N*64\r\n$GPRMC,,V,,,,,,,,,,N*53\r\n$GPVTG,,,,,,,,,N*30\r\n$GPGGA,,,,,,0,00,99.99,,,,,,*48\r\n$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30\r\n$GPGSV,2,1,07,08,,,26,10,,,25,11,,,23,12,,,07*72\r\n$GPGSV,2,2,07,14,,,24,26,,,26,27,,,26*7C\r\n$GPGLL,,,,,,V,N*64\r\n$GPRMC,,V,,,,,,,,,,N*53\r\n$GPVTG,,,,,,,,,N*30\r\n$GPGGA,,,,,,0,00,99.99,,,,,,*48\r\n$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30\r\n$GPGSV,2,1,06,08,,,25,10,,,25,11,,,23,14,,,23*70\r\n$GPGSV,2,2,06,26,,,26,27,,,26*7E\r\n$GPGLL,,,,,,V,N*64\r\n$GPRMC,,V,,,,,,,,,,N*53\r\n$GPVTG,,,,,,,,,N*30\r\n$GPGGA,,,,,,0,00,99.99,,,,,,*48\r\n$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30\r\n$GPGSV,2,1,08,08,,,25,10,,,27,11,,,23,14,,,23*7C\r\n$GPGSV,2,2,08,16,,,26,17,,,30,26,,,28,27,,,26*78\r\n$GPGLL,,,,,,V,N*64\r\n$GPRMC,,V,,,,,,,,,,N*53\r\n$GPVTG,,,,,,,,,N*30\r\n$GPGGA,,,,,,0,00,99.99,,,,,,*48\r\n$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30\r\n$GPGSV,2,1,08,08,,,26,10,,,26,11,,,23,14,,,23*7E\r\n$GPGSV,2,2,08,16,,,25,17,,,26,26,,,27,27,,,26*73\r\n$GPGLL,,,,,,V,N*64\r\n$GPRMC,,V,,,,,,,,,,N*53\r\n$GPVTG,,,,,,,,,N*30\r\n$GPGGA,,,,,,0,00,99.99,,,,,,*48\r\n$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30\r\n$GPGSV,2,1,08,08,,,26,10,,,25,11,,,22,14,,,22*7D\r\n$GPGSV,2,2,08,16,,,25,17,,,25,26,,,26,27,,,26*71\r\n$GPGLL,,,,,
```

nebrius, 2017-06-15:
A thought: is each data chunk suppose to be split per line? All of the data above was from a single data chunk

dtex, 2017-06-15:
@nebrius It should split the messages on \r\n so that's not it. You don't appear to be acquiring a satellite signal. Can you try it near a window?

nebrius, 2017-06-22:
Sorry for the delay, but I finally got around to testing outdoors and the code in https://github.com/nebrius/raspi-io/issues/94#issuecomment-304953213 works for me.

@avcohen can you test with your Raspberry Pi outside?

If you are unsure if your GPS unit has signal or not, check the raw output and look for the `GPGLL` command, and then check the letter in the second to last comma-separated entry. If it says `V`, that means `void`, aka you don't have signal. If it says `A`, that means active and you do have signal.

Let us know if it works!

nebrius, 2017-09-13:
Closing due to inactivity. If this is still an issue, feel free to reopen!

closed 2017-09-13 by nebrius