nebrius/raspi-io #95: Unable to get any movement from servo motors.

josephschenck, 2017-06-22:
I am trying to get a hang of raspi-io so I can use it in an upcoming project, but I am having difficulty getting any sort of movement from my servo motor. 

Whenever I attempt to run the code I have I get the following readout from console but the motor remains static.

> 1498166995719 Available RaspberryPi-IO  
> 1498166996078 Connected RaspberryPi-IO  
> 1498166996092 Repl Initialized  

Here is the code I am running

> var five = require("johnny-five");
> var Raspi = require("raspi-io");
> 
> var board = new five.Board({
>   io: new Raspi()
> });
> 
> board.on("ready", function() {
> 
>   var servo = new five.Servo({
>     pin: "P1-32",
>   });
>   servo.sweep();
> 
> });

dtex, 2017-06-22:
@josephschenck How are you powering the servo? Is there a common ground?

josephschenck, 2017-06-22:
I'm using a breadboard with an external power supply.
[http://i.imgur.com/xMkPe6l.jpg](url)

nebrius, 2017-06-22:
Looking at that photo, it looks like you don't have a common ground. Try connecting a wire from a GND pin on the RPi to the ground rail on your breadboard and that should fix it.

dtex, 2017-09-12:
@josephschenck Just noticed this issue was still open. Did adding a common ground help? Can we close this?

josephschenck, 2017-09-12:
yea, closing it is fine

nebrius, 2017-09-13:
Thanks for following up @dtex and @josephschenck!

closed 2017-09-13 by nebrius