nebrius/raspi-io #97: Install script throws an error

rbwilcoxon11, 2017-09-13:
The install script seems to have a bug in it... I think it could be resolved if the right dependencies were installed.

Here's my package.json

```javascript
{
  ...
  "dependencies": {
   ...
    "raspi-io": "^8.1.0"
  },
  ...
```

When I run

```bash
npm install
```

I get this error back

```
> raspi-soft-pwm@4.0.1 preinstall /home/pi/hue-ball/node_modules/raspi-soft-pwm
> sh ./script/pigpio.sh

pigpio is already installed, skipping installation
npm ERR! Cannot read property 'pause' of undefined
```
However, when I run

```bash
npm install raspi-soft-pwm@5.0.0
```

I don't get the "pause" error. I could get around this by getting all the packages individually and wrapping them like this package does, but...

nebrius, 2017-09-14:
Hi @rbwilcoxon11! I ran into this issue too and documented it over at https://github.com/nebrius/raspi-io/issues/96.

Can you let me know which version of Node.js and npm you're running?

rbwilcoxon11, 2017-09-15:
Node version - v8.5.0
npm version - 5.4.1

Right now i'm getting around it by installing all the dependencies manually and combining them like this files packages src

[image]

The up-carrots are misleading though; each of those is the specific version I installed of each package

nebrius, 2017-09-15:
Can you try downgrading to npm 5.3.0? That's what fixed it for me.

rbwilcoxon11, 2017-09-15:
Works 👍

closed 2017-09-25 by nebrius