nebrius/raspi-io #98: Typescript d.ts

pankleks, 2017-10-15:
Hello,

Is there any support for d.ts file.
I tried: `@types/raspi-io`

but no luck.
Thanks!

nebrius, 2017-10-15:
Interestingly enough, most of Raspi IO is itself written in TypeScript, but I haven't gotten around to porting Raspi IO itself. I haven't created a .d.ts file since I've been meaning to convert it to TypeScript (which changes how I would distribute this).

In the mean time, you can create your own and then include it using the `typeRoots` option in tsconfig. I think it should look something like:

```TypeScript
declare module 'raspi-io' {
  export interface IOptions {
    enableSerial?: boolean;
    enableSoftPwm?: boolean;
    includePins?: Array<number | string>;
    excludePins?: Array<number | string>; 
  }
  function raspi(options?: IOptions): void;
  export = raspi;
}
```

Note: I haven't tested this, so there may be typos and bugs.

pankleks, 2017-12-19:
ok thanks

closed 2017-12-19 by pankleks