nebrius/raspi-io #99: Raspi-io is hanging node red

selfus, 2017-10-22:
I am trying to use  [node-red-contrib-gpio](https://flows.nodered.org/node/node-red-contrib-gpio) that requires raspi-io

However, when installing raspi-io (I am installing it in the .node-red directory), nodered isn't capable to start after a reboot.

Can provide more info if required.

monteslu, 2017-10-22:
Did you install as the pi user and run as root?

selfus, 2017-10-22:
I installed as the pi user, not sure if node-red is running as root though, but it shouldn't

How can I install it as root?

monteslu, 2017-10-22:
raspi-io requires installing as non-root and running as root. Take a look at the notes here:  https://github.com/nebrius/raspi-io#system-requirements

It's likely that you are running node red as the pi user.  You should be able update the systemd script to use the root user.

selfus, 2017-10-22:
Why does raspi-io need to be run as root? Can't this be changed?

Thanks!

nebrius, 2017-10-23:
raspi-io needs to run as root because of the way that Raspbian itself configures GPIO access. In order to read/write or change settings on a GPIO pin (e.g. setting it as input vs output, pull-up resistors, etc), Raspbian requires that the app in question have root access. Unfortunately this is a requirement from the OS.

nebrius, 2017-11-22:
I'm going to close this issue due to inactivity. If you still are running into this issue though, let us know!

closed 2017-11-22 by nebrius

selfus, 2017-12-04:
@nebrius No more issues, thank you for your kind support, and sorry for not closing the issue