nebrius/raspi-io #100: moto-zero

zoobot, 2017-11-07:
Hi
I am trying to translate your the raspberry pi pin scheme into the johnny five motor pin scheme.

With the python GPIO lib it looks like this:
import RPi.GPIO as GPIO
GPIO.setmode(GPIO.BCM)
Motor1A = 27
Motor1B = 24
Motor1Enable = 5 

the Johnny-five motors look like this
var motor3 = new five.Motor({
      pins: {
        dir: 'P1-23',
        cdir: 'P1-16',
        enable: 5,
        invertPWM: true
      }
    });

Do you have any examples of Motors with raspi-io? 
It seems the johnny-five Motor is expecting a number so not sure how it would translate the P1-13 string?

Thanks!!
I'm using this: https://cdn.shopify.com/s/files/1/0176/3274/files/MotoZero_User_Guide_1.2.pdf

nebrius, 2017-11-07:
Hi @zoobot! Can you clarify what you mean when you say that Johnny-Five is expecting a number? Are you getting an exception, or is the motor not turning but otherwise there are no errors, etc?

Also, are you passing `enableSoftPwm: true` to the Raspi IO constructor? `P1-23` and `P1-16` do not support PWM natively, so it needs to be faked in software. IIRC the Python library fakes it for you, but Raspi IO requires that you explicitly enable it by hand.

zoobot, 2017-11-07:
Hi! Thanks for getting back to me!

I am confused about how to translate the pins. 
I am getting this error:

node_modules/raspi-io/node_modules/raspi-io-core/dist/index.js:489
        throw new Error('Unknown pin "' + pin + '"');
        ^
Error: Unknown pin “undefined"

According to the motozero documentation it should be this: https://cdn.shopify.com/s/files/1/0176/3274/files/MotoZero_User_Guide_1.2.pdf <https://cdn.shopify.com/s/files/1/0176/3274/files/MotoZero_User_Guide_1.2.pdf> 
motor1 = Motor(24, 27) 
motor1_enable = OutputDevice(5, initial_value=1) 
motor2 = Motor(6, 22) 
motor2_enable = OutputDevice(17, initial_value=1) 
motor3 = Motor(23, 16) 
motor3_enable = OutputDevice(12, initial_value=1) 
motor4 = Motor(13, 18) 
motor4_enable = OutputDevice(25, initial_value=1)

Translation according to your pin chart would be this
 var motor1 = new five.Motor({
    pins: { dir: ‘P1-35', cdir: ‘P1-36', enable: ‘P1-18, invertPWM: true, enableSoftPwm: true }
  });
  var motor2 = new five.Motor({
    pins: { dir: ‘P1-22', cdir: 'P1-31', enable: 17, invertPWM: true, enableSoftPwm: true } //There is no 17!!
  });
  var motor3 = new five.Motor({
    pins: { dir: ‘P1-33', cdir: ‘P1-10', enable: ‘P1-19', invertPWM: true, enableSoftPwm: true }
  });
  var motor4 = new five.Motor({
    pins: { dir: ‘P1-21', cdir: 18, enable: ‘P1-35', invertPWM: true, enableSoftPwm: true } //There is no 18!!
  });

But I also found this so not sure what is correct. Neither are working:
https://pinout.xyz/pinout/motozero <https://pinout.xyz/pinout/motozero>
 var motor1 = new five.Motor({
    pins: { dir: ‘P1-13', cdir: ‘P1-12', enable: ‘P1-29', invertPWM: true, enableSoftPwm: true }
  });
  var motor2 = new five.Motor({
    pins: { dir: ‘P1-15', cdir: 'P1-31', enable:  ‘P1-11', invertPWM: true, enableSoftPwm: true } 
  });
  var motor3 = new five.Motor({
    pins: { dir: ‘P1-36', cdir: ‘P1-23', enable: ‘P1-32', invertPWM: true, enableSoftPwm: true }
  });
  var motor4 = new five.Motor({
    pins: { dir: ‘P1-12', cdir: ‘P1-33', enable: ‘P1-22', invertPWM: true, enableSoftPwm: true } 
  });

dtex, 2017-11-08:
Re: the Motor constructor expecting a number. Pins can be passed as numbers or strings, it just depends on the situation. These motor examples are for Arduino so they use numbers. Pis address pins differently (a point I'll leave to @nebrius since he's the expert).

I notice that in one of your examples you are missing a single quote and that quote styles throughout are mixed ( ‘ VS ' ). I understand those could just be copy paste problems and don't reflect you actual code but please check.

````
var motor1 = new five.Motor({
    pins: { dir: ‘P1-35', cdir: ‘P1-36', enable: ‘P1-18, invertPWM: true, enableSoftPwm: true }
});
````

Should be:
````
var motor1 = new five.Motor({
    pins: { dir: 'P1-35', cdir: 'P1-36', enable: 'P1-18', invertPWM: true, enableSoftPwm: true }
});
````

zoobot, 2017-11-08:
Thanks for noting those quotes, it is a copy paste email error, not actual code. Pesky quote mismatches!!

This is my test code, have tried a bunch of different pin tests, so far haven’t gotten past the pin error. I'll try to translate Pi pins to this constructor next: new five.Motor([ 3, 12 ]);

const raspi = require('raspi-io');
const five = require('johnny-five');
const board = new five.Board({
  io: new raspi()
});

board.on('ready', () => {
  // Create an Led on pin 7 on header P1 (GPIO4) and strobe it on/off
  // (new five.Led('P1-7')).strobe();
  const led = new five.Led('P1-7');
  led.strobe(500);

  var motor1 = new five.Motor({
    pins: { dir: 'P1-13', cdir: 'P1-12', enable: 'P1-29', invertPWM: true, enableSoftPwm: true }
  });
  var motor2 = new five.Motor({
    pins: { dir: 'P1-15', cdir: 'P1-31', enable:  'P1-11', invertPWM: true, enableSoftPwm: true }
  });
  var motor3 = new five.Motor({
    pins: { dir: 'P1-36', cdir: 'P1-23', enable: 'P1-32', invertPWM: true, enableSoftPwm: true }
  });
  var motor4 = new five.Motor({
    pins: { dir: 'P1-12', cdir: 'P1-33', enable: 'P1-22', invertPWM: true, enableSoftPwm: true }
  });
  motor1.forward(150);
  motor3.reverse(150);
  motor4.reverse(150);
});

nebrius, 2017-11-09:
Hmm, sounds like there's some sort of mismatch going on between Raspi IO and Johnny-Five when doing the pin resolution.

Can you try using the [Pin Information Guide](https://github.com/nebrius/raspi-io/wiki/Pin-Information) to hand-translate the pin to a number and see how it works?

As for the question "why is the pin information so complicated?" The Raspberry Pi makes it complicated unfortunately. The pin number, GPIO number, and other numbers that can be used to describe what we think of as a "pin" are inconsistent between Raspberry Pi versions, unlike the Arduino. Also, pins on the Raspberry Pi can be used for multiple purposes, e.g. `P1-35` can be used for `SPI1` _or_ `PWM1` _or_ `GPIO19`, etc. So for `P1-35`, should the number be 1 or 19? There's no obvious answer, so Raspi IO allows a variety of ways of identifying pins.

zoobot, 2017-11-10:
I’ve tried to hand translate, have tried all different pins with different variations of the constructor.
I think these are probably the correct pins but not sure about how to get them working with the motor constructor AND the enableSoftPwm.

  var motor1 = new five.Motor({
    pins: { dir: 5, cdir: 2, enable: 21, enableSoftPwm: true }
  });

>> node_modules/raspi-io/node_modules/raspi-io-core/dist/index.js:489
        throw new Error('Unknown pin "' + pin + '"');
        ^
Error: Unknown pin "undefined"

var motor1 = new five.Motor([ 5, 2, 21,true]);
>> node_modules/raspi-io/node_modules/raspi-io-core/dist/index.js:489
        throw new Error('Unknown pin "' + pin + '"');
        ^
Error: Unknown pin "true"

var motor1 = new five.Motor([ 5, 2, 21,true]);
>> node_modules/raspi-io/node_modules/raspi-io-core/dist/index.js:539
        throw new Error('Pin "' + pin + '" does not support mode "' + modeName + '"');
        ^
Error: Pin "5" does not support mode “pwm"

dtex, 2017-11-10:
@zoobot You are trying to pass the enableSoftPwm option in the pins object. It should be in the opts object passed to new raspi()

https://github.com/nebrius/raspi-io#new-raspioptions

zoobot, 2017-11-10:
That got me much further!! Thanks! No more errors.
motor1.reverse(200) is working but motor1.forward(200) is not working. 
The motor will turn both ways if I switch the pins in the motor constructor and call motor1.reverse. 

const raspi = require('raspi-io');
const five = require('johnny-five');
const board = new five.Board({
  io: new raspi({ enableSoftPwm: true, includePins: ['GPIO24', 'GPIO27', 'GPIO5'] })
});

board.on('ready', () => {
  //var motor1 = new five.Motor([ 'GPIO27','GPIO24', 'GPIO5']);
  var motor1 = new five.Motor(['GPIO24', 'GPIO27', 'GPIO5']);

  //console.log("Go dog go");
  //motor1.forward(200);
  //motor1.start(255)
  // board.wait(5000, function () {
  //  motor1.stop();
  // });

  console.log("Back dog back");
  motor1.reverse(200);
  board.wait(5000, function() {
    motor1.stop();
  });
});

nebrius, 2017-11-22:
I'm going to close this issue since we're now discussing it in the Johnny-Five repo. Thanks again for filing, and for being patient!

closed 2017-11-22 by nebrius