nebrius/raspi-io #101: Error: Cannot find module 'raspi-io'

ahmadasadi, 2018-02-12:
hi i've installed johnny5 and raspi-io but when i use johnny5 in node-red ,it says "Error: Cannot find module 'raspi-io'" and can not connected.
here is the problem ( pls help me):
Stop Node-RED
pi@raspberrypi:~ $ npm install johnny-five raspi-io
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.

+ johnny-five@0.13.0
+ raspi-io@8.1.0
added 6 packages in 26.429s

pi@raspberrypi:~ $ node-red-start

Start Node-RED

Once Node-RED has started, point a browser at http://192.168.1.5:1880
On Pi Node-RED works better with the Firefox or Chrome browser

Use   node-red-stop                          to stop Node-RED
Use   node-red-start                         to start Node-RED again
Use   node-red-log                           to view the recent log output
Use   sudo systemctl enable nodered.service  to autostart Node-RED at every boot
Use   sudo systemctl disable nodered.service to disable autostart on boot

To find more nodes and example flows - go to http://flows.nodered.org

Starting as a systemd service.
12 Feb 21:09:00 - [warn] [johnny5:5a8b567d.59b738] Error: Cannot find module 'raspi-io'

monteslu, 2018-02-12:
@ahmadasadi while you do install raspi-io as a the pi user, you have to actually run it as the root user:  https://github.com/nebrius/raspi-io#system-requirements

nebrius, 2018-02-12:
I'm admittedly not very familiar with node-red. Where is your node-red project stored at?

Looking at the error message, these lines pop out at me:

```
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
```

This tells me that you tried to run `npm install johnny-five raspi-io` from `/home/pi`, but your project code is located somewhere else. The npm command needs to be run from the root fo your project folder, where the package.json file is located. From this error code, it's implied to me that it's somewhere other than `/home/pi`.

ahmadasadi, 2018-02-12:
sorry i think these lines means that raspi-io has been installed.
am i right?

ahmadasadi, 2018-02-12:
package.json is in pi@raspberrypi:~/.node-red
and
i run npm command as a root user 
in the following
and the result is  these lines:
pi@raspberrypi:~ $ cd .node-red
pi@raspberrypi:~/.node-red $ ls
flows_raspberrypi_cred.json  package.json       settings.js.bak
flows_raspberrypi.json       package-lock.json  settings.js.save
lib                          projects
node_modules                 settings.js
pi@raspberrypi:~/.node-red $ sudo npm install johnny-five raspi-io

> serialport@6.0.5 install /home/pi/.node-red/node_modules/serialport
> prebuild-install || node-gyp rebuild

sh: 1: prebuild-install: not found
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/6.12.3"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/.node-red/node_modules/serialport/.node-gyp"
gyp WARN EACCES user "root" does not have permission to access the dev dir "/home/pi/.node-red/node_modules/serialport/.node-gyp/6.12.3"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/.node-red/node_modules/serialport/.node-gyp"
gyp WARN EACCES user "root" does not have permission to access the dev dir "/home/pi/.node-red/node_modules/serialport/.node-gyp/6.12.3"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/.node-red/node_modules/serialport/.node-gyp"
gyp WARN EACCES user "root" does not have permission to access the dev dir "/home/pi/.node-red/node_modules/serialport/.node-gyp/6.12.3"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/.node-red/node_modules/serialport/.node-gyp"
gyp WARN EACCES user "root" does not have permission to access the dev dir "/home/pi/.node-red/node_modules/serialport/.node-gyp/6.12.3"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/.node-red/node_modules/serialport/.node-gyp"
gyp WARN EACCES user "root" does not have permission to access the dev dir "/home/pi/.node-red/node_modules/serialport/.node-gyp/6.12.3"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/.node-red/node_modules/serialport/.node-gyp"
gyp WARN EACCES user "root" does not have permission to access the dev dir "/home/pi/.node-red/node_modules/serialport/.node-gyp/6.12.3"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/.node-red/node_modules/serialport/.node-gyp"
gyp WARN EACCES user "root" does not have permission to access the dev dir "/home/pi/.node-red/node_modules/serialport/.node-gyp/6.12.3"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/.node-red/node_modules/serialport/.node-gyp"
gyp WARN EACCES user "root" does not have permission to access the dev dir "/home/pi/.node-red/node_modules/serialport/.node-gyp/6.12.3"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/.node-red/node_modules/serialport/.node-gyp"
gyp WARN EACCES user "root" does not have permission to access the dev dir "/home/pi/.node-red/node_modules/serialport/.node-gyp/6.12.3"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/.node-red/node_modules/serialport/.node-gyp"

and above lines results frequently and non stopping

what can i do next ?
thanks a lot

nebrius, 2018-02-13:
Hmm, for some reason that folder is owned by a different user. Can you try running:

```
sudo chmod -R 775  /home/pi/.node-red/
```

ahmadasadi, 2018-02-13:
hi 
my raspbian  version is :+1: 
pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.3 (stretch)
Release:        9.3
is it ok for installing johnny-five and raspi-io?

fivdi, 2018-02-13:
Yes, Raspbian Stretch is ok for Johnny-Five and Raspi-IO.

The output posted above shows the following error:
```
13 Feb 09:47:15 - [warn] [serialport] Error: Module version mismatch. Expected 48, got 14.
```
This means that Node.js v0.11 or v0.12 is being used at runtime but Node.js v6 is expected.

The Node.js module version numbers for the different versions of Node.js can be found at https://nodejs.org/en/download/releases/

Raspi-IO itself requires [Node.js v4.0.0 or higher](https://github.com/nebrius/raspi-io/blob/master/package.json#L33).

It looks like the Raspberry Pi is not configured correctly. Perhaps the users `root` and `pi` are accessing different versions of Node.js.

What's the output of the following two commands?
```
node -v
sudo node -v
```

ahmadasadi, 2018-02-13:
hi
thanks a lot
answer is :+1: 
pi@raspberrypi:~ $ node -v
v6.12.3
pi@raspberrypi:~ $ sudo node -v
[sudo] password for pi:
v6.12.3

ahmadasadi, 2018-02-13:
sorry the thumb sign came by mistake
excuse me
i mean answer is :

fivdi, 2018-02-13:
Hmm, I don't understand what's happening yet.

What's the output of the the following commands?
```
nodejs -v
sudo nodejs -v
```

nebrius, 2018-02-14:
There's definitely some sort of Node.js version mismatch at play, but what I can't say specifically. From the output of `node -v` and `sudo node -v` it looks like pathing is set up correctly there. @monteslu do you know if node-red ships its own Node.js runtime?

Another thing I see: you shouldn't use `sudo` with `npm install`. Instead, you should run `npm install johnny-five raspi-io` without the `sudo` part. Can you try without `sudo` and see if that helps?

nebrius, 2018-04-04:
I'm closing this due to inactivity. Please feel free to reopen if you're still having issues.

closed 2018-04-04 by nebrius