nebrius/raspi-io #102: Developing on PC, deploying to PI

TheBekker, 2018-04-04:
Is it possible to allow the package to install on x86/PC.
Since its only allowed on arm, its not really possible to develop on a pc and just pull a repo from the PI to deploy.

rwaldron, 2018-04-04:
I'm not sure if this meets your needs or even helps get you closer, but I like to use rsub (which is rmate for sublime text), though I'm sure there is an alternative for whichever editor you prefer. 

- https://github.com/henrikpersson/rsub
- https://github.com/textmate/rmate

nebrius, 2018-04-04:
I use rsync myself to copy code from my Laptop to my PC. @rwaldron's method works too.

I think it depends on if you have a build stage or not. I write most of my code in TypeScript, so I always have a command to run before the code can be deployed easily, and I typically chain the two together. If you don't have a build step though, then @rwaldron's method works better.

The command I run is `npm run build && raspi-tools s`, which makes use of the [raspi-tools helper module](https://github.com/nebrius/raspi-tools) I wrote to make working on Raspi IO easier. The rsync command it uses under the hood is:

```bash
rsync -avz -e ssh --exclude node_modules /path/to/repo pi@ip.add.re.ss:/home/pi/repo
```
I then configure Visual Studio Code to remote debug the Raspberry Pi with the following launch.json:

```json
{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "node",
      "request": "attach",
      "name": "Debug RPi",
      "address": "ip.add.re.ss",
      "port": 9229,
      "sourceMaps": true,
      "localRoot": "${workspaceRoot}",
      "remoteRoot": "/home/pi/repo",
      "outFiles": [
        "${workspaceRoot}/dist/**/*.js"
      ]
    }
  ]
}
```

Note: you have to start the app on the Raspberry Pi with the following command:

```Bash
node --inspect-brk=ip.add.re.ss:9229 index.js
```

nebrius, 2018-04-05:
> Or do you just add to package.json and ignore that, since its gonna be build later on the pi?

Yep. I install my dev dependencies (TypeScript, tslint, etc.) with `npm install --only-dev --force`, which ignores cpu warnings and skips raspi-io.

As for raspi-tools, I mostly just mentioned it for the example as I never polished it up for public consumption (no docs as you mentioned). The rsync command I mentioned is what raspi-tools does under the hood.

TheBekker, 2018-04-05:
Ahh that makes sense, didnt know about the effects from those parameters.
Thanks alot :)

rwaldron, 2018-04-05:
This has been very informative for me :D

nebrius, 2018-05-15:
Thanks for the discussion everyone, I'm going to close this issue now since there's no tasks to be done for Raspi IO

closed 2018-05-15 by nebrius