nebrius/raspi-io #104: incompatibility with I2S

alexdufetel, 2018-05-12:
It appears raspi-io can not be run in parallel with a device using I2S on the RPi 3. 
I am using Adafruit's speaker bonnet (https://learn.adafruit.com/adafruit-speaker-bonnet-for-raspberry-pi/overview). When I use raspi-io, the sound coming out of the bonnet is slowed down (seems like a sampling speed issue). 
The bonnet uses GPIO pins 18,19, and 21, which I ignore when setting up the board (otherwise the board does not work at all).

fivdi, 2018-05-13:
This most likely cause of this issue and an explanation of how it can be resolved can be found at https://github.com/joan2937/pigpio/issues/87.

nebrius, 2018-05-14:
Thanks for the report, and sorry we couldn't help more. I added a note to the README documenting this limitation.

nebrius, 2018-05-14:
You might try something like https://learn.adafruit.com/usb-audio-cards-with-a-raspberry-pi/instructions, which should get around this limitation.

fivdi, 2018-05-14:
@nebrius In theory it should be possible to resolve this issue. The pigpio Node modules allows you to configure whether the PCM or PWM clock is used for "software" PWM with the [configureClock](https://github.com/fivdi/pigpio/blob/master/doc/configuration.md#configureclockmicroseconds-peripheral) method.

If the Raspi-IO constructor provided an option for specifying which clock to use then either:
```js
pigpio.configureClock(5, pigpio.CLOCK_PCM);
```
or:
```js
pigpio.configureClock(5, pigpio.CLOCK_PWM);
```
could be called based on that option.

If the PWM clock is used I2S should work.

That's the theory. I don't have the required hardware to test this to see if it works in practice.

nebrius, 2018-05-14:
@fivdi I can do the test since I've got plenty of RPis and an oscilloscope, although it'll...uhh...be quite a while due to travel. I'll keep this issue open as a reminder to look into it in early June.

alexdufetel, 2018-05-15:
thanks for the updates - if you need help testing a fix out happy to

darweshsinghcgi, 2018-08-08:
@nebrius Any luck with the testing?

nebrius, 2018-08-11:
@darweshsinghcgi ack, sorry, this slipped through the cracks and I haven't even started yet 😓 

I'll try and get to it soon, before JSConf US in a little over a week.

darweshsinghcgi, 2018-08-17:
I'm not sure how to test it but I have a RPi, I2S mic, and piezo on order...let me know if I can be of any help.

nebrius, 2018-09-12:
So I _finally_ got around to testing this...and...changing the clock mode to PWM causes raspi-pwm to crash as soon as it calls `hardwarePwmWrite`. @fivdi, it would appear that PWM mode makes it impossible to use the hardware PWM for PWM output, does this match up with your understanding?

```
Error: pigpio error -100 in gpioHardwarePWM
    at Gpio.hardwarePwmWrite (/home/pi/raspi-test/node_modules/pigpio/pigpio.js:90:12)
    at PWM.write (/home/pi/raspi-test/node_modules/raspi-pwm/dist/index.js:131:19)
    at RaspiIOCore.pwmWrite (/home/pi/raspi-test/node_modules/raspi-io-core/dist/index.js:598:30)
    at RaspiIOCore.analogWrite (/home/pi/raspi-test/node_modules/raspi-io-core/dist/index.js:589:12)
    at Led.value (/home/pi/raspi-test/node_modules/johnny-five/lib/led/led.js:113:19)
    at Led.value (/home/pi/raspi-test/node_modules/johnny-five/lib/led/led.js:100:14)
    at Led.brightness (/home/pi/raspi-test/node_modules/johnny-five/lib/led/led.js:272:8)
    at pwmTest (/home/pi/raspi-test/src/index.js:64:7)
    at Board.<anonymous> (/home/pi/raspi-test/src/index.js:123:11)
    at <anonymous>
```

fivdi, 2018-09-12:
> it would appear that PWM mode makes it impossible to use the hardware PWM for PWM output, does this match up with your understanding?

@nebrius yes, this is the case. The pigpio C library documents this restriction [here](http://abyz.me.uk/rpi/pigpio/pdif2.html#hardware_PWM). Unfortunately the pigpio Node.js module doesn't document the restriction.

nebrius, 2018-09-12:
Thanks. I have an idea for fixing this, but it'll require some re-architecting, so stay tuned folks.

What I'm thinking now is that I'll change the behavior of `enableSoftPwm` so that it _only_ uses software based PWM and ignores hardware based PWM entirely, even on pins that support it. Then, I can remove all the hardware/software PWM logic from raspi-io-core and add some (much simpler) PWM logic to raspi-io. This'll actually be nice, since it'll make raspi-io-core simpler and more platform independent. It'll take some time though, so a little more patience everyone.

nebrius, 2018-09-13:
Thinking about this more, I really should do https://github.com/nebrius/raspi-io-core/issues/6 first before tackling this issue to make sure there aren't other regressions.

nebrius, 2019-02-21:
Whew, that took a long time, but the rewrite of Raspi IO is complete, and a fix for this is (theoretically) in! Can you test out Raspi IO v10.0.1 to verify? Note that there are some breaking changes with v10.

nebrius, 2019-04-17:
I'm closing this issue as it seems the problem has been resolved. Please reopen if you are still experiencing issues.

closed 2019-04-17 by nebrius