nebrius/raspi-io #105: Not compatible with RaspiRobot V3f?

robertjarske, 2018-08-17:
Hello!
I'm using raspi-io together with johnny-five for a pi rover project. I encountered an issue when trying to setup the motors.
According to the python example I got with the rover kit the following pins should be used:
    
```
    RIGHT_PWM_PIN = 14
    RIGHT_1_PIN = 10
    RIGHT_2_PIN = 25
    LEFT_PWM_PIN = 24
    LEFT_1_PIN = 17
    LEFT_2_PIN = 4
    SW1_PIN = 11
    SW2_PIN = 9
    LED1_PIN = 8
    LED2_PIN = 7
    OC1_PIN = 22
    OC2_PIN = 27
    OC2_PIN_R1 = 21
    OC2_PIN_R2 = 27
    TRIGGER_PIN = 18
    ECHO_PIN = 23 
```

My code:

```const raspi = require('raspi-io');
const five = require('johnny-five');
const board = new five.Board({
  io: new raspi({enableSoftPwm: true})
});

board.on('ready', function() {

var left = new five.Motor(['GPIO24', 'GPIO4', 'GPIO17']);
var right = new five.Motor(['GPIO14', 'GPIO25', 'GPIO10']);

  console.log("Start");
  left.forward(55);
  right.forward(55);

  board.wait(2000, () => {
   left.reverse(55);
   right.reverse(55);
  });

  board.wait(4000, () => {
   left.stop();
   right.stop();
  });
});
```

The error I get is that GPIO14 does not support mode pwm, when I try the "left" motor it works as intended. Any help at all would be much appreciated!

nebrius, 2018-08-17:
Try setting [enableSoftPwm](https://github.com/nebrius/raspi-io/#new-raspioptions) to true in the constructor and that will fix it.

The Raspberry Pi doesn't actually support PWM on GPIO14, but it's possible to emulate PWM in software. It comes at a bit of a cost though: it increases CPU usage and it's not quite as accurate. I came to a different conclusion on how surface this information to the user than the author's of the Python library, as I think it's important for the user to be aware of this.

robertjarske, 2018-08-17:
```
const raspi = require('raspi-io');
const five = require('johnny-five');
const board = new five.Board({
  io: new raspi({enableSoftPwm: true})
});

board.on('ready', function() {

var left = new five.Motor(['GPIO24', 'GPIO4', 'GPIO17']);
var right = new five.Motor(['GPIO14', 'GPIO25', 'GPIO10']);

  console.log("Start");
  left.forward(55);
  right.forward(55);

  board.wait(2000, () => {
   left.reverse(55);
   right.reverse(55);
  });

  board.wait(4000, () => {
   left.stop();
   right.stop();
  });
});``` 

I do that already I think in the "new raspi({enableSoftPwm: true})" ? Or should it be somewhere else? Still no luck, also tried having it inside the "new five.Motor" but it won't work.

What puzzles me is that one of the motors are working as intended but not the other?

nebrius, 2018-08-17:
Oh whoops, sorry I totally missed you already had `enableSoftPwm`. Hmm, OK yeah that's strange. `enableSoftPwm` only works in the `raspi` constructor.

Can you give me the full stack trace of the exception?

fivdi, 2018-08-17:
According to the changelog GPIO14 and GPIO15 are not available with raspi-io@6.0.0 or higher. See https://github.com/nebrius/raspi-io/blob/master/CHANGELOG.md#600-2016-3-20.

nebrius, 2018-08-17:
Oh RIGHT! Thanks @fivdi, I forgot my own work there for a second :D

nebrius, 2018-08-18:
Ohhhh, I see the issue now. GPIO14 is reserved for use with the serial port. If you move it to a different pin it should work

robertjarske, 2018-08-20:
The motor board is not soldered but can only be plugged into the raspberry pi in one way so I guess you could call it hard wired. 

[image]
[image]

robertjarske, 2018-08-22:
@nebrius sorry to be a pain but do you have any updates regarding this issue? Anything I can do?

nebrius, 2018-09-06:
@robertjarske sorry for the delay, I just got back from Burning Man so I've been off the grid for a long time.

I actually don't know if this will work cause I don't really remember this part of the implementation, but in an ideal world it will work: try passing `enableSerial: false` to the Raspi IO constructor and see if it starts working. If not, then I should add support for that.

robertjarske, 2018-09-07:
@nebrius I tried setting enableSerial to false but unfortunately I get the same error  as before

nebrius, 2018-09-07:
@robertjarske Thanks for trying that. I'll leave this open so I can investigate getting this behavior changed. I can't promise I can make this change (there may be unintended side-effects of setting a pin mode on a serial port pin after the app is closed), but this sounds like a good change to make.

robertjarske, 2018-09-08:
@nebrius thank you for looking into it

nebrius, 2019-02-21:
@robertjarske I know this is considerably late, I'm sorry for that. I just finished a major rewrite I started around the time of my last comment. Now it has logic to not filter out the serial pins if `enableSerial` is set to `false`. It should work for you now.

nebrius, 2019-02-21:
I'm going to go ahead and close this issue as the immediate problem is fixed, but if you run into any other problems please reopen and we'll work out the next step.

closed 2019-02-21 by nebrius