nebrius/raspi-io #106: Create SPI straw-proposal

nebrius, 2018-08-20:
As part of getting SPI support into Johnny-five, create a straw-proposal for SPI in [raspi-io-core](https://github.com/nebrius/raspi-io-core) based on your initial work in [raspi-spi](https://github.com/nebrius/raspi-spi).

nebrius, 2018-09-13:
Thinking on this one some more, I think this task is blocked by https://github.com/nebrius/raspi-io/issues/107, in practice

geo242, 2019-05-27:
Now that #107 is closed, can this be revisited? I have tried using [pi-spi](https://github.com/natevw/pi-spi) while using raspi-io and it seems there is a conflict. SPI stops working as soon as raspi-io is initialized.

nebrius, 2019-05-27:
I’m swamped with other projects unfortunately, so I don’t have time to work on this right now. That said, you can use the `excludePins` constructor option to exclude the SPI pins, allowing you to use pi-spi

geo242, 2019-05-27:
@nebrius I did try that, but I used the include pins instead and just included the one pin I needed and that didn't help. I'll try the exclude and see if that operates differently.

nebrius, 2019-06-05:
@geo242 thanks for being patient, I've been traveling. That's unusual, that should have worked. Can you show me the code you're running? I wonder if there's a bug in `includePins`

nebrius, 2021-01-17:
Not going to work on this since this project is in maintenance mode

closed 2021-01-17 by nebrius