nebrius/raspi-io #109: Integration with INA219

msi-matthew, 2018-09-10:
I am trying to build a Raspberry Pi controller that switches a pin state based on the voltage, and was hoping to use the following modules to make this work together.

https://www.npmjs.com/package/ina219
https://www.npmjs.com/package/raspi-io
https://www.npmjs.com/package/johnny-five

I have successfully read the current and voltage from an INA219 connected to a Raspberry Pi, and also turned an LED on/off with raspi-io and johnny-five.

Based on feedback from @dtex (https://github.com/rwaldron/johnny-five/issues/1500#issuecomment-419728752), I was wondering if the GPIO pins had to be declared in any way to get these modules working together.  @dtex thought the modules (INA219 & raspi-io) may need shared access to the GPIO pins. Ideally the INA219 would use a 3.3V, ground, and both I2C pins, and johnny-five / raspi-io would use the rest of them.

Is this possible with how the raspi-io module is written? 

Is this a good approach to this problem?

nebrius, 2018-09-13:
Pins cannot be shared, that's fundamentally the way that microcontrollers work. Pins often support more than one type of functionality (e.g. pin `P1-3` supports the SDA pin of I2C, GPIO Input, and GPIO Output), but they can only be used for one of these at a time, not both.

I'm not sure this is actually your question though. Can you confirm that what you want to do is use the `ina219` library linked on `P1-3` and `P1-5`, while having an LED connected to an entirely different pin, e.g. `P1-7`, is this correct?

If so, yes Raspi IO and ina219 should work together just fine without modification.

msi-matthew, 2018-09-13:
Yes, the idea is to use pins PI-1, 3, 5, 9 for the INA module, and some of the other pins 7, 11, 13, etc. for controller switching.

nebrius, 2018-09-13:
Oh interesting, what are pins 5 and 9 being used for in the INA module? (they're not I2C pins)

msi-matthew, 2018-09-13:
I'm using the PI indexing scheme. The breakout board needs I2C as well as power and ground.

1 - 3.3V Power
3 - SDA (I2C)
5 - SCL (I2C)
9 - Ground

[image]

nebrius, 2018-09-13:
Oh, lol, right...that's what I get for not checking my own pinout diagrams 😅

So yes, you should be able to use that module without interference from Raspi IO, and you should not need to do any special configuration to get them to work together. Everything should _just work_ AFAIK. (famous last words?)

msi-matthew, 2018-09-13:
I did get some of it working, but ran into memory issues. I'm trying again with the `rpio` module to do some switching instead of `johnny-five` + `raspi-io`.

Here's the thread with the memory issues if you are interested: https://github.com/rwaldron/johnny-five/issues/1504#issuecomment-421041814

And the `rpio` module: https://www.npmjs.com/package/rpio

nebrius, 2018-09-13:
Since it sounds like this is now focused solely on memory issues and we have a separate issue opened for it, I'm going to go ahead and close this issue.

closed 2018-09-13 by nebrius