nebrius/raspi-io #111: Very high CPU usage on Raspberry Pi

skpulipaka26, 2019-01-12:
[image]

Guys
Is this normal? 
The Pi becomes totally unresponsive after a few minutes of the usage.

nebrius, 2019-01-14:
Hmmm, I haven't seen that behavior myself, especially when nothing is set to INPUT mode (which admittedly does cause a bit of a CPU spike). Can you let me know which Raspberry Pi you have and which version of Raspbian you're running? I'll try my best to reproduce this behavior.

P.S. Please use a word other than "guys" to refer to groups of developers. Not all developers and contributors to this project are men.

skpulipaka26, 2019-01-15:
Noted and updated the original post. 
 
Thank you very much.

nebrius, 2019-01-15:
Thanks for the updated info @skpulipaka26! Hmmm, I'm especially surprised your CPU load is so high on the 3 B+. I'm out and about for the evening, but I'll check it out tomorrow morning (PST). This definitely shouldn't be happening, thanks for letting me know!

nebrius, 2019-01-15:
I test everything out, and indeed I'm seeing the same behavior 😱 It might take me a while to figure out what's going on, but this is definitely a regression _somewhere_. Sorry you're having issues, I'll do my best to get this fixed quickly.

skpulipaka26, 2019-01-15:
@nebrius 

Thank you for addressing this issue.

nebrius, 2019-01-15:
Welp, that went faster than I thought and led to a dead-end faster than I thought. Turns out, it's some native code... _somewhere_ ...that's causing it, as we can see in this profiler trace (`(program)` means native code).

[image]

@fivdi do you have any ideas off the top of your head? Don't worry about digging if you don't have any immediate ideas.

FWIW I'm currently in the middle of a rewrite of Raspi IO, and one of the things I'm doing is relying on cached values a lot more than before. This _hopefully_ will fix the issue so we aren't calling into C++ nearly as often.

skpulipaka26, 2019-01-15:
I am sorry, I closed the issue by mistake.

I have reopened it again

skpulipaka26, 2019-01-15:
@nebrius 
I was testing more stuff on the Pi. The CPU stayed around 1% when I was using onoff library. I am technically not aware of differences, but Johnny Five library better suits my purpose.

nebrius, 2019-01-15:
I'm going to reopen this issue. Even if I don't get to this issue soon, I'd like to make sure it stays on my radar.

nebrius, 2019-01-15:
Using Johnny-Five won't ever be at 1% like onoff because it does _significantly_ more than onoff, at least 2 orders of magnitude more stuff. That said, it shouldn't be above 25% at idle in the LED example, at worst.

skpulipaka26, 2019-01-15:
Makes sense. That justifies the 70+ % CPU usage by some native code.

fivdi, 2019-01-15:
> @fivdi do you have any ideas off the top of your head? Don't worry about digging if you don't have any immediate ideas.

I can't dig in at the moment as I'm no where near a Pi. The CPU usage numbers above show that the program spent 72.62% of its time in native code. Two important pieces of information are missing here. What native code is being called and how often is it being called. The example program blinks an LED at 1Hz and only two native calls per second should be needed to achieve this. Assuming that raspi-io is using either Gpio#digitalWrite or Gpio#pwmWrite (part of pigpio) to blink the LED then I would expect raspi-io to spend very little time in pigpio native code.

@nebrius Do you know what native methods are being called and how often they are being called? I would imagine that rasp-io is making a lot of native calls. Perhaps even tens of thousands of native calls per second. The last time I checked it was possible to call Gpio#digitalWrite approximately 1.4 million times per second in a Raspberry Pi 3. I don't know how often Gpio#pwmWrite can be called per second.

> Using Johnny-Five won't ever be at 1% like onoff because it does significantly more than onoff, at least 2 orders of magnitude more stuff. That said, it shouldn't be above 25% at idle in the LED example, at worst.

I would expect a JohnnyFive + BeagleBone-IO program that blinks an LED at 1Hz on a BeagleBone Black to require about 1% CPU time. BeagleBone-IO uses onoff for digital I/O.

fivdi, 2019-01-16:
> Currently, I do know that digital reads are expensive because of the constant "call the handler and emit and event" pattern it uses.

@nebrius I wouldn't expect the LED blink program posted above to result in digital read being called.

> Under the hood it uses a setInterval currently, but I am going to switch it to be all event based, which should hopefully lighten that load.

If the input pin is floating this may increase the load.

>  I'll have to keep digging, but it's not going to be easy.

Perhaps the easiest thing to do is to add `console.log` statements to methods like `digitalWrite`, `pwmWrite` and `hardwarePwmWrite` in https://github.com/fivdi/pigpio/blob/master/pigpio.js to see if they are being called too often.

nebrius, 2019-01-17:
Some good news! I managed to get the [rewrite of Raspi IO](https://github.com/nebrius/core-io/tree/typescript) to a functional state! It only supports GPIO so far, but CPU usage with this new architecture sits at only 6.9% once the JIT has fully engaged.

[image]

I'll be sure to keep testing performance as the rewrite progresses, but I think we can safely say that the rewrite and caching it employs will fix this issue.

skpulipaka26, 2019-01-17:
That's wonderful news. 
Can I test it too?

nebrius, 2019-01-17:
Sure! There's a couple hoops to jump through, but running these steps in this order _should_ work (untested though):

1. Remove 'raspi-io' from the dependencies section of your `package.json` file
1. Delete your `package-lock.json` file and you're `node_modules` folder
1. Run `npm install git://github.com/nebrius/core-io.git#typescript`
1. Run `npm install git://github.com/nebrius/raspi-io.git#typescript`
1. Run `npm install` to install the rest of your dependencies

fivdi, 2019-01-17:
> Some good news! I managed to get the rewrite of Raspi IO to a functional state! It only supports GPIO so far, but CPU usage with this new architecture sits at only 6.9% once the JIT has fully engaged.

Yes, that is good news. 6.9% CPU usage for hello world on a Raspberry Pi 3 sounds about right. Most of that 6.9% will be the pigpio C library sampling GPIO 0-31 250,000 times per second in the background. It's not possible to prevent the pigpio C library from doing this.

justincathcart, 2019-02-01:
I was having the same issue as @skpulipaka26 with high CPU loads on the Raspberry Pi 3 B + and a memory leak. Running top showed the CPU at 130% and memory steadily increasing until crashing with:

`462907 ms: Scavenge 693.7 (723.1) -> 692.2 (727.1) MB, 16.3 / 0.0 ms  (average mu = 0.177, current mu = 0.119) allocation failure

FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory`

I took the steps that @nebrius noted and things worked fine afterward. I too am showing CPU at 6.9% and memory holding steady at 3.9 MB. One thing to note for anyone else troubleshooting is that the new version of raspi-io no longer has a default export. So you have to update `const Raspi = require(raspi-io)` to `const Raspi = require('raspiio').RaspiIO` or `import { RaspiIO as Raspi } from 'raspi-io'`

nebrius, 2019-02-01:
Ah right, thanks for reminding me about the export change @justincathcart. I have a note to update the documentation in the development branch, but haven't gotten around to it yet.

justincathcart, 2019-02-01:
Thanks @nebrius for your amazing work on this and other open source projects! It's much appreciated.

sandeep45, 2019-02-05:
@nebrius using the #typescript branch does solve the CPU and out of memory issue but now I am unable to do the HelloWorld LED Test on certain pins like Pin GPIO19. The error says: `Error: pwmWrite is not supported by Raspi IO`

If I change the pin number to GPIO4, then it works fine. This issue does not exist in the master branch and seems to only happen in the `typescript` branch. Bellow is the entire error stacktrace and code used.

```
pi@raspberrypi:~/code/johnny_play $ sudo `which node` led_test.js
1549389561253 Available Raspi IO
1549389561264 Connected Raspi IO
1549389561275 Repl Initialized
>> /home/pi/code/johnny_play/node_modules/abstract-io/dist/index.js:99
        throw new Error(`pwmWrite is not supported by ${this.name}`);
        ^

Error: pwmWrite is not supported by Raspi IO
    at CoreIO.pwmWrite (/home/pi/code/johnny_play/node_modules/abstract-io/dist/index.js:99:15)
    at CoreIO.analogWrite (/home/pi/code/johnny_play/node_modules/abstract-io/dist/index.js:239:14)
    at Led.value (/home/pi/code/johnny_play/node_modules/johnny-five/lib/led/led.js:113:19)
    at Led.value (/home/pi/code/johnny_play/node_modules/johnny-five/lib/led/led.js:100:14)
    at Led.on (/home/pi/code/johnny_play/node_modules/johnny-five/lib/led/led.js:236:8)
    at Board.board.on (/home/pi/code/johnny_play/led_test.js:12:7)
    at Board.emit (events.js:202:15)
    at Board.EventEmitter.emit (domain.js:439:20)
    at processTicksAndRejections (internal/process/next_tick.js:74:9)
    at process.runNextTicks [as _tickCallback] (internal/process/next_tick.js:51:3)

```

```
const Raspi = require('raspi-io').RaspiIO;
const five = require('johnny-five');

const board = new five.Board({
  io: new Raspi(),
  repl: true,
  debug: true
});

board.on('ready', () => {
  const led = new five.Led('GPIO19');
  led.on();
  board.repl.inject({led: led});
});
```

nebrius, 2019-02-05:
Hrm, that's kinda but not exactly expected. I haven't implemented support for PWM in the new branch yet, so I'd expect anything that tries to do a PWM write would fail, and the LED class does make use of PWM.

Now the question, why is J5 trying to do a PWM write in some cases but not others? @dtex, do you have any thoughts?

nebrius, 2019-02-10:
FYI PWM and I2C are now working! Serial is working too and passes unit tests, but I haven't verified in my integration tests yet running on actual hardware yet.

nebrius, 2019-02-21:
Version 10.0.0 was just released! I verified that CPU usage is indeed low. The most I saw it spike to on an RPi 3 was around 40% with a lot of streaming data coming from a temperature sensor (as well as software PWM running in the background).

Let me know if this fixes CPU usage for everyone, and then I think we can mark this issue as resolved!

nebrius, 2019-04-17:
I'm closing this issue as it seems the problem has been resolved. Please reopen if you are still experiencing issues.

closed 2019-04-17 by nebrius