nebrius/raspi-io #112: raspi-io detecting "OTHER" board on Pi 3

bennytheshap, 2019-01-23:
I'm running into an issue where raspi-io cannot detect the board type of my Pi 3 (with up-to-date Raspbian). The detected board type is "OTHER" and there are 0 pins mapped. 

Code is here: https://github.com/CUBoulder-2019Sp-IML4HCI/Wonkavator

Any ideas what might be wrong?

Thank you

nebrius, 2019-01-23:
Hi @bennytheshap. There are two things at play here, at least one of which is on me to fix.

First, can you copy paste the entire warning message printed to the console saying `Unknown board revision ${rev}, assuming Raspberry Pi Zero/2/3 pinout.`? I need to know what `rev` is equal to so I can add it to the list.

Second though: when Raspi IO encounters a board revision that it doesn't know about, it's _supposed_ to map to the RPi 2/3/Zero pinout automatically. Can you describe in more detail what you mean by `0 pins mapped`? Do you mean that `five.io.pins` is an empty array, or something else?

bennytheshap, 2019-01-23:
That's part of what's odd. There is no warning message. But pins is an empty array.

Here's what I get:
/home/ben/devel/wonkavator/node_modules/raspi-io-core/dist/index.js:505
        throw new Error('Unknown pin "' + pin + '"');
              ^
Error: Unknown pin "25"
    at RaspiIOCore.value (/home/ben/devel/wonkavator/node_modules/raspi-io-core/dist/index.js:505:15)
    at RaspiIOCore.digitalWrite (/home/ben/devel/wonkavator/node_modules/raspi-io-core/dist/index.js:631:45)
    at Relay.off (/home/ben/devel/wonkavator/node_modules/johnny-five/lib/relay.js:78:11)
    at J5Relay.off (/home/ben/devel/wonkavator/src/wonkavator.ts:33:18)
    at new J5Relay (/home/ben/devel/wonkavator/src/wonkavator.ts:24:14)
    at Object.<anonymous> (/home/ben/devel/wonkavator/src/wonkavator.ts:47:20)
    at Module._compile (internal/modules/cjs/loader.js:721:30)
    at Module.m._compile (/home/ben/devel/wonkavator/node_modules/ts-node/src/index.ts:439:23)
    at Module._extensions..js (internal/modules/cjs/loader.js:732:10)
    at Object.require.extensions.(anonymous function) [as .ts] (/home/ben/devel/wonkavator/node_modules/ts-node/src/index.ts:442:12)
[nodemon] app crashed - waiting for file changes before starting...

nebrius, 2019-01-23:
Uhhh, that’s unexpected 😅

Can you run the following command and copy paste the output to this issue?

```
cat /proc/cpuinfo
```

bennytheshap, 2019-01-23:
Here you go. 

Do you have a few minutes for a video chat? I could demo some weirdness that might be helpful.

```
[ben@wonkavator:~/devel/wonkavator $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a22082
Serial		: 000000002c4f1e60](url)
```

bennytheshap, 2019-01-23:
I think I got it working. Turns out one should not try to reference a pin before the board is ready, as the mappings don't yet exist.

closed 2019-01-23 by bennytheshap

nebrius, 2019-01-24:
> I think I got it working. Turns out one should not try to reference a pin before the board is ready, as the mappings don't yet exist. Thanks for your assistance.

Ah yes, that would do it. You can't do anything with the API until the ready event has fired. Glad you figured it out!