nebrius/raspi-io #113: Does not support mode "servo"

ThadeuLuz, 2019-02-18:
/home/pi/qwee/node_modules/raspi-io-core/dist/index.js:546
        throw new Error('Pin "' + pin + '" does not support mode "' + modeName + '"');
        ^

Error: Pin "4" does not support mode "servo"
    at RaspiIOCore.value (/home/pi/qwee/node_modules/raspi-io-core/dist/index.js:546:15)
    at RaspiIOCore.pinMode (/home/pi/qwee/node_modules/raspi-io-core/dist/index.js:512:21)
    at Servo.value (/home/pi/qwee/node_modules/johnny-five/lib/servo.js:58:19)
    at new Servo (/home/pi/qwee/node_modules/johnny-five/lib/servo.js:199:8)
    at Board.<anonymous> (/home/pi/qwee/src/s.js:9:17)
    at Board.emit (events.js:194:15)
    at Board.EventEmitter.emit (domain.js:441:20)
    at process._tickCallback (internal/process/next_tick.js:61:11)
    at Function.Module.runMain (internal/modules/cjs/loader.js:745:11)
    at startup (internal/bootstrap/node.js:283:19)

ThadeuLuz, 2019-02-18:
I'll try to figure this out and I'll re-open if needed.

closed 2019-02-18 by ThadeuLuz

nebrius, 2019-02-19:
Glad you got it sorted out!