nebrius/raspi-io #116: Reboot and permission requirements

hegelstad, 2019-04-30:
Requiring a reboot before start and requiring raspi-io to be installed as a user but ran as a root makes it very difficult to use raspi-io with dockerized builds such as through BalenaCloud.

https://github.com/fivdi/pi-io doesn't require a reboot so why is this needed with raspi-io?

nebrius, 2019-05-09:
My apologies for the delay, I've been on vacation.

The reboot is required to enable I2C on the device. Note: if you enable I2C on your own and reboot it yourself before installing raspi-io, then you do not need to reboot again. If you do not use I2C, you can ignore this warning. pi-io assumes that you have already configured your device for I2C use manually, which is why it doesn't print a warning. If you try and use pi-io to access I2C without enabling I2C first, you'll get a cryptic error.

This is not my own requirement, it's set by Raspbian itself. Unfortunately this means that I don't have any control over it.

hegelstad, 2019-05-09:
Ah thank you. So I guess pi-io's documentation ideally would make a remark about having to manually enable I2C, and perhaps raspi-io's documentation should state that I2C only needs to be enabled, if so by installing and rebooting or manually?

Thanks for responding anyways.

closed 2019-05-09 by hegelstad

nebrius, 2019-05-09:
> raspi-io's documentation should state that I2C only needs to be enabled, if so by installing and rebooting or manually?

[Like I mention here](https://github.com/nebrius/raspi-io#i2c-notes)?  😉

hegelstad, 2019-05-10:
Hmm, I see that it is mentioned, but it seems like it is a hard requirement to do the reebot to be able to use any of the features:

> Note: You must reboot your Raspberry Pi the first time you install Raspi IO on it!

> 
> raspi-io's documentation should state that I2C only needs to be enabled, if so by installing and rebooting or manually?
> 
> Like I mention here? 😉