nebrius/raspi-io #117: GPIO inconsistent behavior

d1runberg, 2019-06-01:
When using the basic blink example the LED blinks for a few seconds, then acts sporadic and then starts blinking again. If I switch to just the .on() method it acts the same way; will be on for a few seconds and then turn off / start to blink and then just be off.

nebrius, 2019-06-22:
@d1runberg can you provide some example code demonstrating this behavior, and let me know which Raspberry Pi you're running on?

d1runberg, 2019-06-23:
Hi @nebrius ,
I am running a pi ZeroW V1.1
Node Version: 10.16.0

Here is my code:

```const { RaspiIO } = require('raspi-io');
const five = require('johnny-five');
const board = new five.Board({
  io: new RaspiIO()
});

board.on('ready', () => {
  // Create an Led on pin 7 on header P1 (GPIO4) and strobe it on/off
  const led = new five.Led('P1-7');
  led.strobe(500);
});```

Thank you for helping me out on this!

nebrius, 2019-06-25:
I tested this on my end, and I don't see the behavior you're seeing. Some details on my test setup:

- OS: 2019-06-20-raspbian-buster-lite
- Node: v10.16.0, downloaded from https://unofficial-builds.nodejs.org/download/release/v10.16.0/
- Raspi IO v10.0.2

With this setup, I only see about 10% CPU usage:

[image]

Which OS are you running, and which version of Raspi IO are you running?

nebrius, 2021-01-17:
Closing due to inactivity

closed 2021-01-17 by nebrius