nebrius/raspi-io #118: Raspi is not a constructor

Akil313, 2019-06-06:
Using a Raspberry Pi 3 Model B+
Installed Johnny-Five and Raspi-io

After installing, I ran this using node

```
const Raspi = require("raspi-io");
const five = require("johnny-five");

const board = five.Board(
    {
        io: new Raspi()
    }
);

board.on("ready", () => {
    let led = new five.Led('GPIO18');

    this.repl.inject({
        redLed: led
    });
})
```

After running with node, I get the error Raspi is not a constructor.
I've used the default code from the Johnny Five site and still no luck.

Akil313, 2019-06-06:
After identifying the difference between the Raspberry Pi implementation from the Johnny Five website and this repo, I was able to get the code to run properly.
Specifically the change was `const Raspi = require("raspi-io");` to `const Raspi = require("raspi-io").RaspiIO;`

closed 2019-06-06 by Akil313

dtex, 2019-06-07:
@Akil313 Was [this](https://github.com/rwaldron/johnny-five/blob/d2edcbe0557bdf4b91801f9b7f2121348eb4e4d2/docs/raspi-io.md) the page?

Akil313, 2019-06-07:
Yea that's the one. Tried coping that exact code and ran into the error.

nebrius, 2019-06-07:
Ah right, I forgot to update the code samples there when I did the big refactor. Thanks for looking into it @dtex. I'm still recovering from JSConf EU 😅