nebrius/raspi-io #119: Unknown board revision c03111

jackcannon, 2019-07-06:
Using a Raspberry Pi 4 Model B
Installed Johnny-Five and Raspi-io

Error:
```
Unknown board revision c03111, assuming Raspberry Pi Zero/2/3 pinout. Unless you are running a compute module or very old RPi this should work fine. Please report this board revision in a GitHub issue at https://github.com/nebrius/raspi-board.
```

```
const five = require("johnny-five");
const Raspi = require("raspi-io").RaspiIO;
const board = new five.Board({
  io: new Raspi()
});

board.on("ready", () => {
  console.log('ready');
  // Create an Led on pin 13
  const led = new five.Led(13);

  led.blink(500);
});
```

jackcannon, 2019-07-06:
Having seen issue https://github.com/nebrius/raspi-io/issues/49, here is the output requested in that issue:

```
$ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2835
Revision	: c03111
Serial		: 10000000f1ce20cd
```

nebrius, 2019-07-10:
Thanks for the report @jackcannon! I'm still waiting to get mine delivered, so I hadn't been able to add support yet.

I just published an update to `raspi-board`, the dependency that handles board revision detection. The warning you're seeing should be gone now if you run `npm update`.

nebrius, 2019-09-10:
This issue should be fixed now, so I'm going to close this issue. Thanks again for the report!

closed 2019-09-10 by nebrius