nebrius/raspi-io #121: PCA9685 and HT16K33 Not Working Together

becauseinterwebs, 2019-08-06:
Hey Nebrius!

I posted this over at the Johnny Five github and dtex says J5 checks out, so wondering if there "might" be something here.

I have the [PCA9685 16 channel I2C interface board](https://www.adafruit.com/product/815) and the [8x8 led matrix](https://www.adafruit.com/product/870) (using HT16K33) from Adafruit.  I cannot get both of them working together.

The PCA9685 is controlling an RGB Led and I have it switching colors every second.  Whenever I do anything with the 8x8 (HT16K33), the LED just freezes at the current color and does nothing.  The 8x8 led works fine in all cases.

The address of the PCA9685 is 0x40 and the LED matrix is at 0x70.  I'm running this on a Raspberry Pi Zero W using Raspi-IO.  

I've tried connecting the 8x8 to the pass-thru headers of the PCA9685 and also running the SDA/SCL (as well as VCC and GND) into a bread board and then hooking the 8x8 and PCA up to the breadboard.  Same results.

It seems the HT16K33 is blocking the PCA9685 somehow?

Here is the code:

```
const Raspi = require('raspi-io').RaspiIO;
const five = require('johnny-five');
const board = new five.Board({
    io: new Raspi()
});

// Heart shape for 8x8
var heart = [
    "01100110",
    "10011001",
    "10000001",
    "10000001",
    "01000010",
    "00100100",
    "00011000",
    "00000000"
  ];

 var matrix = new five.Led.Matrix({
    addresses: [0x70],
    controller: "HT16K33",
    rotation: 4,
  });

// index for color switching
let index = 0;
let rainbow = ["FF0000", "FF7F00", "FFFF00", "00FF00", "0000FF", "4B0082", "8F00FF"];

// setup the RGB connected to the pins on the PCA9685
let rgb = new five.Led.RGB({
    pins: {
        red: 0,
        green: 1,
        blue:  2
    },
   addresses :  [0x40], // with or without this line doesn't seem to matter
    controller : "PCA9685"
});

// set color of RGB LED and strobe at random interval
const setColor = () => {
    rgb.color(rainbow[index]);
    rgb.strobe(Math.floor(Math.random() * 2000));  
    index++;
    if (index === rainbow.length) {
        index = 0;
    }
    setTimeout(setColor, 2000);
}

const startMatrix = () => {
  matrix.clear(); // led freezes here
  matrix.draw(heart);
}

board.on('ready', () => {
    setColor();
    // using a 10 second delay during which time the Led blinks just fine
   setTimeout(startMatrix, 10000); 
});

board.on('exit', () => {
    // When the board disconnects, turn the LEDs off.
    rgb.stop().off();
    matrix.clear().off();
    console.log('[johnny-five] Board closing - bye bye!');
});
```

dtex, 2019-08-06:
@nebrius I don’t have all the hardware, so when I tested I just monitored the I2C writes in firmata.js and all looked good.

becauseinterwebs, 2019-08-07:
I've been digging into the libraries tonight.  After the matrix is cleared and writes out, the blink method of the RGB is still be called.  I have traced all the way into the raspi-i2c and j5-io libraries and the i2cwrite method is still being called (I just have the led blinking red at this point):

i2cWrite(address, registerOrInBytes, inBytes)

output:
i2cwrite 64 [ 6, 0, 0, 4096, 16 ] undefined
i2cwrite 64 [ 10, 0, 0, 4096, 16 ] undefined
i2cwrite 64 [ 14, 0, 0, 4096, 16 ] undefined

i2cwrite 64 [ 6, 4096, 16, 0, 0 ] undefined
i2cwrite 64 [ 10, 0, 0, 4096, 16 ] undefined
i2cwrite 64 [ 14, 0, 0, 4096, 16 ] undefined

So it's definitely still calling the methods.  I also added a matrix.clear().off() after it being on for 2 seconds and it clears and goes off, but the RGB is still unresponsive.  I also have the HT16K33 (matrix) attached to the pass-through pins of the PCA9685 now and I can control the matrix without any issues.

nebrius, 2019-08-07:
At a super quick glance, I don't see any obvious reason why it shouldn't be working. Unfortunately I'm _suuuupppper_ swamped right now and won't have any free time to look into this until September 😱.

One thing you might try, just to rule things out, is running this code on an Arduino to verify that it is indeed something RPi specific.

becauseinterwebs, 2019-08-08:
Ok tested using an arduino with a PCA9685 connected to it with an led and a HT16K33 chained to it and getting same results so moving back over to J5 thread.

Thank you!

becauseinterwebs, 2019-08-08:
Seems to be issue in J5

closed 2019-08-08 by becauseinterwebs