nebrius/raspi-io #122: High CPU Usage on Raspberry PI 4

sergei202, 2019-09-02:
I'm having the exact same issue of 100% CPU load as #111 but with a Raspberry Pi 4B:

- Node v10.16.3
- Raspbian 10/Buster
- raspi-io v10.0.2

[image]

Is there anything else I can provide or test?

Thanks for this awesome project!

nebrius, 2019-09-03:
Thanks for letting me know @sergei202.

A couple of things to try:
- Can you try quitting VSCode, since it is also running Node.js?
- Can you try running with `debug: false` in the board constructor?
- Can you let me know which version of pigpio you're running?

Last time I checked, Raspi IO was completely broken on the RPi4 due to pigpio being broken down the line, but it looks like it's running ok on your board. I just got back from Burning Man, and I'm in the process of unpacking, but I'll try and take a peek soon.

sergei202, 2019-09-04:
@nebrius: How was Burning Man??

I should have done a better job describing my setup/screenshot:
- I'm running VSCode locally on a laptop
- The code is running on a Pi 4 (mounted via sshfs)
- The 2 terminal windows are just ssh sessions
- Running the code with `debug: false` didn't help
- I'm running v71 of of pigpiod and v1.2.3 of pigpio

I've actually gotten everything to run on the Pi 4 using Raspi IO except for some pixel LED strips!
The only problem is that it just pegs out the CPU and runs out of memory after a few minutes :sweat_smile: 

Here's the exact code I'm running and executing on the Pi with `sudo ts-node test.ts`:
```ts
import { Board, Led }    from 'johnny-five';
import { RaspiIO }       from 'raspi-io';

const board = new Board({
    repl: false,
    debug: false,
    io: new RaspiIO()
});

board.on('ready', () => {
    console.log('Board ready.');
    var boardBlue = new Led({pin:'GPIO5'});
    boardBlue.blink(250);
});
```

fivdi, 2019-09-08:
This issue appears to be related to the serial port which raspi-io enables by default. If the serial port is enabled CPU usage is 100% and more (a Pi 4 has 4 cores.) If the serial port is disabled CPU usage goes back to normal.

@sergei202 Is CPU usage normal with the following test code?
```js
import { Board, Led }    from 'johnny-five';
import { RaspiIO }       from 'raspi-io';

const board = new Board({
    repl: false,
    debug: false,
    io: new RaspiIO({enableSerial:false})
});

board.on('ready', () => {
    console.log('Board ready.');
    var boardBlue = new Led({pin:'GPIO5'});
    boardBlue.blink(250);
});
```

sergei202, 2019-09-09:
@fivdi: I'm out of town until next week, but I'll test it as soon as I'm back!

srib1997, 2019-09-10:
> This issue appears to be related to the serial port which raspi-io enables by default. If the serial port is enabled CPU usage is 100% and more (a Pi 4 has 4 cores.) If the serial port is disabled [...]

It solved my problem
Thank you.

nebrius, 2019-09-10:
Weeeeeeird, `enableSerial` is supposed to default to `false` on the RPi 4. See https://github.com/nebrius/raspi-io/blob/master/src/index.ts#L69-L75.

I think I'm finally going to have time to look into it today and tomorrow. I'll see what I can reproduce.

fivdi, 2019-09-11:
> Weeeeeeird, `enableSerial` is supposed to default to `false` on the RPi 4. See https://github.com/nebrius/raspi-io/blob/master/src/index.ts#L69-L75.

I'd expect this code to set `enableSerial` to `true` on a Pi 4.

sergei202, 2019-09-17:
@nebrius: That fixed it!  Thank you so much!

closed 2019-09-17 by sergei202