nebrius/raspi-io #124: Add support to Raspberry PI 4

spade69, 2019-10-16:
Add support to Raspberry PI 4, pls, i tried ,  Add support to Raspberry PI 4
`const rapi = new Raspi();  ` is not working ,  program just halt there, never get into the ready event.  Probably it is not supported yet ?

nebrius, 2019-10-17:
Try running `sudo apt update && sudo apt upgrade`. It's a known bug that all but the most recent version of pigpio (a system library) cause this bug. Also make sure you're running the most up to date version of Raspi IO, as I fixed another bug related to the RPi 4 recently.

spade69, 2019-10-18:
> Try running `sudo apt update && sudo apt upgrade`. It's a known bug that all but the most recent version of pigpio (a system library) cause this bug. Also make sure you're running the most up to date version of Raspi IO, as I fixed another bug related to the RPi 4 recently.

seems like  my raspbian got problems, now i reinstall my raspbian to latest version 2019-09-26 raspbian buster、rasp-io 10.0.4 , and it just works.

closed 2019-10-18 by spade69