nebrius/raspi-io #125: Question on using Johnny-Five and raspi-io to read from a DHT12 Temperature Sensor

microagent, 2019-10-23:
Hello, I read through all the documentation and examples but it isn't clear to be how or if this can be done, by saying hooking the DHT12 sensor (with built in resistor) to Power, Ground, and P1-7. Is this possible and if so, how would the code example look like? I know my basic set is correct a I followed the code example  to get the LED blinking, but that is the only example I can find specific to Raspberry Pi. 

I have been looking here.

Multi - DHT11_I2C_NANO_BACKPACK
http://johnny-five.io/examples/multi-DHT11_I2C_NANO_BACKPACK/

Any help or pointers to perform this specific task? I would be most grateful.

Thank You,
Alan

nebrius, 2019-10-24:
I've never tried to do this before, but a quick scan of the Johnny-Five documentation indicates that the DHT series can _only_ be used with a backpack. @dtex, does this match your understanding?

microagent, 2019-10-25:
Thanks nebrius. Worse case, I found another NPM package that works very well, I had liked the idea of having other sensor options that Johnny-Five presented as my personal project evolved, https://www.npmjs.com/package/node-dht-sensor. 

Thanks so much for posting that video on Johnny-Five and the incredible Wiki, https://github.com/nebrius/raspi-io/wiki/Getting-a-Raspberry-Pi-ready-for-NodeBots, it was extremely useful and the top result by far, after all my searching. 

You have a gift for conveying information to others!

closed 2020-05-08 by nebrius