nebrius/raspi-io #126: Unknown Board Revision c03112 - Raspberry PI 4 with 4GB

D1no, 2019-11-24:
Hi, I try to run Johnny Five on my new Raspberry PI 4 `c03112`. #119 mentioned a new revision number — apparently there is another one. How can I force it in the mean time to use `c03111`?

```bash
pi@raspberrypi:~/servo_pointer $ yarn start
yarn run v1.19.1
$ sudo node ./server.js
Unknown board revision c03112, assuming Raspberry Pi Zero/2/3 pinout. Unless you are running a compute module or very old RPi this should work fine. Please report this board revision in a GitHub issue at https://github.com/nebrius/raspi-board.
1574610392096 Available Raspi IO  
1574610392100 Connected Raspi IO  
1574610392106 Repl Initialized  
>> /home/pi/servo_pointer/node_modules/j5-io/dist/core.js:69
        throw new Error(`Unknown pin "${pin}"`);
        ^

```

```js
const five = require("johnny-five");
const Raspi = require("raspi-io").RaspiIO;

const board = five.Board({
  io: new Raspi()
});

board.on("ready", () => {
  const led = new five.Led("P1-P12");
  led.blink();
});
```

nebrius, 2019-12-04:
Hi @D1no. The message printed is a warning only, and should work just fine with revision `c03111`. I'll get it added to the list.

The crash your experiencing is unrelated to the warning. The issue you're experiencing is because `P1-P12` is not a valid pin. Did you mean `P1-12`?

D1no, 2019-12-16:
Ok I'll need to check that when I play around again. In the last version I had locally I needed to patch add the revision number to the node module.

closed 2019-12-16 by D1no