nebrius/raspi-io #128: raspi-serial v6 is out: please update

WebReflection, 2020-01-10:
Current raspi-io does not build on NodeJS 13 but it does on LTS 12.

The reason is that raspi-serial used serialport v7, which doesn't build on NodeJS 13:
https://github.com/nebrius/raspi-serial/issues/10

It would be nice to update raspi-serial to v6, which uses latest serialport, which builds just fine on NodeJS 13.

I might file a MR instead, but I wanted to touch base first.

The platform I am using/testing is ArchLinux, either 32bit or 64bit.

nebrius, 2020-01-10:
D'oh, I new I was forgetting something! Thanks for the heads up. I just published v11 of raspi-io with the update.

WebReflection, 2020-01-10:
will double check once I can verify all is good with NodeJS 13, and re-open if needed.

meanwhile ... thanks a lot 👍

closed 2020-01-10 by WebReflection