nebrius/raspi-io #130: Instructions for ARMv6 node install are outdated

lafiosca, 2020-05-04:
Hello. I didn't know if it would be acceptable to try to edit the wiki page directly out of the blue, so I thought I'd open an issue about this instead. I am setting up an old Pi (1) Model B I had lying around, and I was following your node installation instructions from the wiki page:

https://github.com/nebrius/raspi-io/wiki/Getting-a-Raspberry-Pi-ready-for-NodeBots

When I went to the node downloads page, I found that the LTS no longer provides an ARMv6 download. I had to find the node 10.x binary listing here to get an ARMv6 version: https://nodejs.org/dist/latest-v10.x/

Additionally there is a typo of "you're version" instead of "your version" in the warning at the end of that section.

Thank you!

nebrius, 2020-05-08:
Thanks for noticing the discrepancy and the typo! I'll go and edit the wiki page, since I believe it's locked down so that only project collaborators can edit, and GitHub sadly doesn't support pull requests on wikis.

FYI there _are_ still unofficial builds created by the Node.js project at https://unofficial-builds.nodejs.org/download/release/ where you can download Node.js 12, 13, and 14 for ARMv6.

nebrius, 2020-05-08:
And done. Thanks for writing in!

closed 2020-05-08 by nebrius

lafiosca, 2020-05-08:
@nebrius I wasn't sure if the wiki was locked down because it shows me an edit button and allows me to navigate to the edit form. I was afraid to hit the Save button and test 😅

nebrius, 2020-05-18:
Oh interesting, turns out I _hadn't_ locked down editing the wiki. Thanks for letting me know!