nebrius/raspi-io #131: L298 motor board receives start signal on script start, cannot stop motor

baggachipz, 2020-05-18:
I've asked around and searched high and low and can't find any answer to my issue; the only piece in here which may be causing the problem is Raspi-IO. I have a Raspberry Pi Zero W, a L298N Board for running a motor, and a simple DC motor. I've simplified my code to cause the issue:

```javascript 
const raspi = require('raspi-io').RaspiIO
const five = require('johnny-five')
const board = new five.Board({
  io: new raspi()
})

board.on('ready', function () {  
  const motor = new five.Motor(['GPIO23', 'GPIO24'])
})
```

Library versions:

```
Raspbian Stretch
NodeJS: v10.15.0
"johnny-five": "^1.4.0"
"raspi-io": "^11.0.0"
```

And the wiring layout is as follows:
[image]

The line `new five.Motor()` causes the motor to start spinning, and it won't stop even if I kill the script. I have swapped out cables, changed the GPIO pins used to three different places, and tried the second output on the L298 board, so I know it's not a hardware issue there. 

I guess that means that the PWM pin (GPIO23 in this example) is set to HIGH, but I'm not exactly sure as I'm new to this. I apologize for being somewhat vague, this is my first foray into this. Could it be that the new version of the library (v11) has some problem here? I feel like I had this working with an older version of raspi-io. Your help is greatly appreciated.

dtex, 2020-05-18:
For your motor initialization try:

````js
const motor = new five.Motor({
  pins: ['GPIO23', 'GPIO24'],
  invertPWM: true
})
````

This [note](http://johnny-five.io/api/motor/#situations-where-invertpwm-true-is-required) explains what that does and why it's necessary sometimes.

baggachipz, 2020-05-18:
Oh my god it worked. Thank you!

closed 2020-05-18 by baggachipz