nebrius/raspi-io #132: Adafruit Motor HAT control board & current sense

oligriffiths, 2020-05-19:
Hi

I am using https://www.adafruit.com/product/2348 and it's working great.

I notice in the Johnny Five docs that some boards can do current sensing http://johnny-five.io/examples/motor-current/

Anyone know if it's possible to do this with this board?

Thanks

Oli

dtex, 2020-05-19:
This one?

https://www.adafruit.com/product/2348

It doesn't look like it. Honestly, the only shield/hat I know of with it built-in is the Arduino Motor Shield R3:

https://store.arduino.cc/usa/arduino-motor-shield-rev3

oligriffiths, 2020-05-19:
Ah ok thanks, I thought I'd ask.
I'm motorising my blinds and wondered if I could get away with not using limit switches. I'm trying with timers and it roughly works but I think over time they'll get out of sync.

dtex, 2020-05-19:
First, that's a brilliant project and I hope you'll share details once it's done.

Second, I've worked with motors a lot and I think this is a job more suited to steppers. That's a bummer because stepper support isn't really there in raspi-io, but you've got options. The steppers could be connected to an I2C backpack running accelStepper. Those backpacks would be addressable and you could control several blinds with a single Pi.

oligriffiths, 2020-05-19:
Yeah i looked into stepper options but I struggled to find a stepper motor with enough torque and not too much current draw to bust the Adafruit hat.
I think probably limiter switches at the bottom will be the simplest option. The motors are working well and able to move the blind, but there is some drift that I am trying to figure out where it's coming from which is why I am thinking that a simple limiter switch might work.

Reading about it, I see various articles suggesting pull up/down resisters are necessary, but others saying that GPIO inputs are protected with a high value resister anyway and that just connecting between the pin an 3.3V would be fine.

Struggling to find info on rasp-io and switches on GPIO pins. Any advice?

oligriffiths, 2020-05-19:
Turns out I was reading the docs wrong and the switches and buttons "just work" http://johnny-five.io/api/switch/ http://johnny-five.io/api/button/ if you address correctly to the GPIO pins.

I also wonder if the reason timing isn't working right is because I am using the motor a lot during my testing and it's getting warmer and thus affecting the torque, usually the blinds will be opened/closed from a cold start so I am testing that now to see if it can be consistent from a cold start.

The other option is potentially accelerating the motor up to full speed rather than going from stopped to full speed instantly, might have a more predictable curve. Will investigate.