nebrius/raspi-io #134: Using Atlas Scientific Ezo PH EC and DO sensors with raspi.io ?

Bohemus307, 2020-09-24:
Does Raspi.io currently support Atlas Scientific Ezo sensors un the Sensor class? I have them wired on an analog to digital converter so they plug into the usb port currently. I also am wondering if you gather sensor data thru the usb port as well with raspi.io? Any whelp with this would be appreciated I would really like to use Rapi.io for this project if possible.

Link to Atlas board i am using 
https://atlas-scientific.com/ezo-accessories/basic-usb-to-serial-converter/

Link to embedded circuits
https://atlas-scientific.com/circuits/

Link to Atlas code they provide
https://www.atlas-scientific.com/files/pi_sample_code.pdf

nebrius, 2020-10-09:
Sorry, this issue slipped through the cracks for me. 

The short answer is I don't know if those sensors are supported or not.

The long answer is it depends on how those devices work. Raspi IO acts like an intermediary between the Operating System and Johnny-Five.

You'll need to figure out how the USB device represents itself to the OS. Does the USB device appear as a serial port or I2C port? Or is it a generic USB device that doesn't emulate one of the other part types? If it's the former, it might work with Raspi IO. If it's the latter, it won't work.

Once you've determined how the device and OS interact, then you'll need to go to the [Johnny Five website](http://johnny-five.io/) to see if it's on the list of supported sensors. If so, then there should be some example code to show you how to get it to work.

Bohemus307, 2020-10-09:
Hey, no worries on time thanks for your reply. I've been digging more into this issue and the device seems to represent itself as a serial port communicating through UART. I am not certain of this but I have managed to get their python example code working and have attempted to use Johnny FIve to communicate with it but so far to no avail. I am not certain if it is just the design of the embedded sensor or if I'm not connecting to the serial port using Johnny-Five correctly. I'm honestly working on a solution using the Python code through a PyShell but was really hoping to have a javascript solution as that's part of the project's overall goal. Either way ill keep digging but if you have any advice or more questions id love any help I can get. Thanks again.

nebrius, 2020-10-09:
This is a good start. Are the sensors your interested in using listed as being supported on the website? To find this out, you'll want to visit the [API page for the type of sensor you want to use](http://johnny-five.io/api/).

Bohemus307, 2020-10-09:
They unfortunately do not. They are atlas scientific embedded circuits and might require some conversion to get them to work with J5 aka Firmata. I'm uncertain of how to go about this but have pinged the j5 crew as well and maybe with some direction and my own sweat equity ill succeed in getting them working. The python code provided is pretty robust so I feel like it is possible and would benefit both the javascript community and Atlas Scientific. I might try pinging them as well for assistance. If you'd like to look into more to help me out the sensor examples are here. https://github.com/Bohemus307/Raspberry-Pi-sample-code
thanks again for your help and for Raspi.io I think its awesome.

nebrius, 2020-10-14:
So the next step would be to add support for the sensor to J5. The way J5 is architected, you can't use sensors that aren't supported, and there's no way to create an adapter to use J5 "off the shelf" for a sensor it doesn't support.

J5 is mostly inactive these days, and we rarely add new features to it anymore sadly. If you're willing to put in the work to add support yourself and submit a PR, I bet we can muster up some folks to review the PR. I wanna make sure I set expectations properly though: you'll probably be on your own for a lot of this work.

Bohemus307, 2020-10-14:
Well sounds daunting but I do enjoy some hard work. My biggest question is how do I go about learning the process of adding this sensor to J5? Is there documentation or would I just be converting python code in some way too Firmata? I am definitely new to this so I apologize for my confusion.

nebrius, 2020-10-16:
To the best of my knowledge there isn't really any documentation, so you'll probably need to dig into the J5 codebase to figure it out.

Bohemus307, 2020-10-16:
Ok lol sounds like it might be over my head but ill see what I can do. Thanks again.

closed 2022-07-11 by Bohemus307