nebrius/raspi-io #135: Installing Raspi-IO To Mac for Purposes of Later Build/Deploy To Pi

hotsparklab, 2020-11-01:
Hello! I'm diving down the rabbit hold of controlling NewPixels directly or indirectly from a Pi 4, with a Node app being deployed via balena.io/docker. The challenge is that I've been working on this app from my Mac, then deploying to a Pi. It's preventing a typical install Raspi-IO that rightfully doesn't like my Mac. I could add dependencies to package.json without install, yet TypeScript complains about such things that aren't installed and can't be referenced locally.

As this is likely an edge case for now, any thoughts on how to bypass the architecture check? Would I fork the repository and disable that while I work? Is there a need to add mock support, allowing installation on other platforms for development purposes?

Thanks for your thoughts in advance. :)

nebrius, 2020-11-01:
You can bypass the architecture check with `npm install --force. You'll still get some install errors from when it tries to install pigpio, but I don't _think_ that will actually halt the npm install.

Alternatively, you can download the declaration files manually and stick them in a "typings" folder that you point "typeRoots" to in your tsconfig file.

hotsparklab, 2020-11-02:
Hello @nebrius, thanks for your assist again on this. I hadn't thought of copying the declaration files. I'm thinking I'll use that and/or switch to npm. I've been using yarn for React frontend/backend for consistency, not required though and seems to be wandering off the beaten path server-side. Worst case because the build process is also off the beaten path, I'll install that module as part of the build process and not immediately, locally (maybe, I have options though!). We can go ahead and close this. Have a great day. 😄

closed 2020-11-02 by hotsparklab