nebrius/raspi-io #139: Installing package on non-arm/raspberry-pi

oligriffiths, 2021-11-04:
Hi

Can someone explain why this package *has* to be installed on the raspberry pi? I want to be able to commit the package to my git repo but can't install it on my macbook.

Thanks

Oli

nebrius, 2021-11-04:
It relies on the C++ [PiGPIO](https://github.com/joan2937/pigpio), which depends on system libraries only available on the Raspberry Pi.

You can get around this with `npm install --force`, however if you try and run the code it will crash on startup.

oligriffiths, 2021-11-04:
Thanks, I only want to checkout the code on my local machine and not run it. So long as it doesn't require any libraries on the pi to actually install then the `--force` command should work. 
Testing this with `yarn install --force` doesn't work however, do you know if it's a different command for yarn?

nebrius, 2021-11-04:
> So long as it doesn't require any libraries on the pi to actually install then the `--force` command should work.

IIRC the pigpio npm module needs to compile the pigpio library on install, so I think it actually does require these libraries on install.

> Testing this with yarn install --force doesn't work however, do you know if it's a different command for yarn?

No idea I'm afraid.

oligriffiths, 2021-11-04:
Ah, is this why the PI needs to be rebooted after install? Because it does some kind of out-of-band install on the device?

nebrius, 2021-11-04:
> Ah, is this why the PI needs to be rebooted after install? Because it does some kind of out-of-band install on the device?

That's one of a couple of reasons. It also makes a few system configuration changes as well to `/boot`. It only needs to reboot if this is the first time installing it on the device.

oligriffiths, 2021-11-04:
Gotcha. I'm just writing some initial setup instructions and wanted to see if I could get this package committed to the repo to make things easier, but it sounds like it needs to be setup on the device to ensure these out-of-band changes are made, unless there is some script that can be run to do this...

nebrius, 2021-11-04:
> Gotcha. I'm just writing some initial setup instructions and wanted to see if I could get this package committed to the repo to make things easier, but it sounds like it needs to be setup on the [...]

Yes, it has to be installed via npm/yarn on the device. Generally speaking, it's bad practice to commit `node_modules` to a repo anyways, especially when native modules are in play (as is the case here) cause they may end up being built for the wrong target/architecture (different Pis have different targets/architectures, so a binary built for one won't necessarily run on another)

oligriffiths, 2021-11-04:
> Generally speaking, it's bad practice to commit node_modules to a repo anyways

yeah I know, I just like to do that for packages marked as not maintained in case they go away in the future.

nebrius, 2021-11-04:
> > Generally speaking, it's bad practice to commit node_modules to a repo anyways
> 
> yeah I know, I just like to do that for packages marked as not maintained in case they go away in the future.

FWIW, packages older than 72 hours can't be deleted from npm

oligriffiths, 2021-11-04:
Oh, good to know, thank you.

fivdi, 2021-11-07:
> IIRC the pigpio npm module needs to compile the pigpio C++ library on install, so I think it actually does require these libraries on install.

It should be possible to install the pigpio npm module successfully on any machine that supports Node.js. For example, here is what I see when I install it on an Intel based Linux machine:
```
$ node -v
v16.9.1

$ npm i pigpio

added 4 packages, and audited 5 packages in 8s

found 0 vulnerabilities
$ 
```

The pigpio npm module doesn't compile the [pigpio C library](https://github.com/joan2937/pigpio) on install. The binaries generated when the pigpio npm module is installed on a non-Pi machine will of course be completely and utterly useless at runtime. This "feature" was added to the pigpio npm module to enable developers to develop on non-Pi systems.