nebrius/raspi-io #140: segmentation fault loading raspi-i2c

dkossman, 2023-03-06:
hello,

hoping someone can point me in the right direction to get past this.  I had a working node server on a Pi Zero W but the SD card died, and i had my source code backed up, but didn't have an image backup... it has been a couple years so not sure what versions of everything i was running.

I'm NOW running the latest Raspberry Pi OS (lite) on a Pi Zero W with the latest patches
- using nodejs 19.7.0

My nodejs "hello world" app works fine.

i've installed the following npm modules:
├── @influxdata/influxdb-client@1.33.2
├── body-parser@1.20.2
├── ds18b20-raspi@0.3.6
├── express@4.18.2
├── influx@5.9.3
├── os@0.1.2
├── path@0.12.7
├── pug@3.0.2
├── raspi-i2c@6.2.4
├── raspi@6.0.1
└── systeminformation@5.17.12

....
const raspi = require('raspi');  // this works
const I2C = require('raspi-i2c').I2C;  // segmentation fault!

can anyone suggest a next step to figure out how to get raspi-i2c to load?

thanks!

dkossman, 2023-03-06:
never mind - "npm rebuild" solved the problem.

closed 2023-03-06 by dkossman

nebrius, 2023-03-06:
Glad you got it figured out.